Xinyuan-LilyGO/LilyGO-T-ETH-Series #1: Missing schematic

zebu1er, 2020-08-15:
It's very necessary to know about what GPIOs are connected with what (pullup, led, IC, ...) when using the board.
Most of the boards maker supply it, is it possible to have it too ?

diasarmando, 2020-08-18:
I need also the schematic.

CrazyRs6, 2020-09-05:
I also am looking for a schematic. Especially the clock lines

kymodoke, 2020-09-15:
+1. I've got five of these boards and I really need the schematic!

CrazyRs6, 2020-12-29:
@kymodoke @diasarmando @zebu1er 

I have been working on this module for a while now, and finally it seems to be working. I have followed the schematic from waveshare: 
https://www.waveshare.com/w/upload/0/08/LAN8720-ETH-Board-Schematic.pdf

I hope this helps!

zebu1er, 2020-12-29:
@CrazyRs6
Thanks !

closed 2023-02-12 by lewisxhe