Xinyuan-LilyGO/LilyGO-T-ETH-Series #3: Clock mode setting GPIO17_OUT does not work with 1.0+ version of the board.

xaaron, 2020-09-27:
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT does not work with my version of the board.  I have to set the clock mode to:  #define ETH_CLK_MODE  ETH_CLOCK_GPIO0_OUT

If I make the above change, this code works on my board.  Do the different version of the board work differently or is this a  mistake in the provided ino?

peterus, 2020-09-29:
On my v1.2 board i have seen the same issue! I debugged multiple days to get the board working...
Thank you for nothing @lewisxhe. The documentation is like always very bad from LilyGO. If they could fix finally there docs from there project they would be a very good hardware seller... but with this: it is **really** hard and **difficult**...

lewisxhe, 2020-09-30:
Sorry, I don’t know the specific version number. I only know that LIilyGo with PSRAM has been discontinued and replaced with the version without PSRAM

kymodoke, 2020-09-30:
> Sorry, I don’t know the specific version number. I only know that LIilyGo with PSRAM has been discontinued and replaced with the version without PSRAM

This strange, as what you say would imply that product versioning numbers are either wrong or they go in decreasing order...

- V**1.0** has a ESP32-**WROOM** without PSRAM
- V**1.2** has a ESP32-**WROVER-B** with 8MB of PSRAM (from which 4MB can be easily be accessible from Arduino sketch).

Mine is v1.2 and it does have PSRAM (verified).
[image]

The

lewisxhe, 2020-10-04:
Yes, the version number is a mystery, is the problem solved?  Please close the problem if it is resolved.  I have added a comment on the PSRAM version in the example.

xaaron, 2020-10-04:
I have version 1.2 and the GPIO0_OUT needs to be set for Ethernet to work.
With the above code change everything works fine.  This is a good board,
but people have to be able to get it to work if it is to be successful in
the market.  I would like to see more people use this board because it
provides the functionality I've been looking for at a good price.

dok-net, 2022-04-12:
It would be interesting to get feedback from users of this board, especially versions other that 1.0 and 1.1, which I have tested, for https://github.com/espressif/arduino-esp32/pull/6527, which is now in arduino-esp32 master.

closed 2023-02-12 by lewisxhe