Xinyuan-LilyGO/LilyGO-T-ETH-Series #4: 5V pin for connection of sensors

fromCologne, 2020-10-28:
Is there a 5V pin or a soldering point where I can access 5V?
A published schematic would be great.

andyboeh, 2021-05-26:
Even with the published schematic, I'm unsure if it's possible to tap 5V somewhere in PoE mode - any ideas?

huub66, 2021-11-22:
anyone found a way to connect 5v sensors to this board?

kymodoke, 2021-11-22:
That is what Logic Level Converters are made for: https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide

kymodoke, 2021-11-22:
On the schematics https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/schematic/T-ETH-POE.pdf there are some components referred to 5V (SY8089AAAC and WPM2015), but I don't know if that comes from USB or from PoE source.

closed 2023-02-12 by lewisxhe

andyboeh, 2023-02-12:
@lewisxhe What is the solution?

lewisxhe, 2023-02-12:
[image]
5V source is connected with VBUS and POE, but no external plug is reserved

andyboeh, 2023-02-12:
Thanks for the clarification, that really is the solution :)

kimble4, 2023-02-12:
And it's eminently practical to solder a header pin or wire to D5, which is large and accessible on the top side of the board.

lewisxhe, 2023-02-12:
Indeed, this is not considered in the design of hardware. If there is a revised version later, I will make suggestions to the hardware design

jamesgohl, 2023-06-17:
> And it's eminently practical to solder a header pin or wire to D5, which is large and accessible on the top side of the board.

My thoughts exactly.  There are 2 * 3v3 pins on the 2*8 header.

Would this be a viable modification?

[image]

Whilst I am no electronic engineer, I am learning. Have I got the correct MOSFET from the schematic, where the source pin connects to the D5 line?

hn, 2023-07-18:
This is how I grabbed 5V from the cathode of diode D5 (SS34). The cable was fixed with some epoxy glue to the ethernet connector. Please do not complain about the soldering quality, I only had a flame thrower at hand.

[image]

victorclaessen, 2024-07-11:
> This is how I grabbed 5V from the cathode of diode D5 (SS34). The cable was fixed with some epoxy glue to the ethernet connector. Please do not complain about the soldering quality, I only had a flame thrower at hand.

Thank you that was very useful!