Xinyuan-LilyGO/LilyGO-T-ETH-Series #5: Board dimensions

looxonline, 2020-11-04:
Hi, I would like to create a 3D printable enclosure for this board. I have just ordered it and it should be arriving in about 15 days but I would like to design the enclosure in the meanwhile. Please could you upload dimensions of the board and the height of the components off the board?

Thanks.

cyberkeiler, 2021-05-27:
There is a tinkercad file by Stefan Nikolaus. He wrote an article (german!) about this board on his website [www.nikolaus-lueneburg.de](https://www.nikolaus-lueneburg.de/2020/07/ttgo-t-internet-poe/)

The File linked in the comments:
[https://www.tinkercad.com/things/knnUYK5V0f5](https://www.tinkercad.com/things/knnUYK5V0f5)

looxonline, 2021-05-27:
> There is a tinkercad file by Stefan Nikolaus. He wrote an article (german!) about this board on his website [www.nikolaus-lueneburg.de](https://www.nikolaus-lueneburg.de/2020/07/ttgo-t-internet-poe/)
> 
> The File linked in the comments:
> https://www.tinkercad.com/things/knnUYK5V0f5

Thanks for the heads up. I ended up designing the board in fusion 360.

greenmagics, 2021-11-12:
> Thanks for the heads up. I ended up designing the board in fusion 360.

Could I ask you to share this file?
Preferably in the form of Fusion360 and STEP.
Thank you in advance

closed 2023-02-12 by lewisxhe