Xinyuan-LilyGO/LilyGO-T-ETH-Series #8: Can't flash the board

bangadrum, 2021-02-23:
Using the proper programmer daughter board, and data LEDs flashing on that board so connection seems to be fine, but no combination of RST/IO0 is letting me flash the board. 
I've tried ESPHome Flasher and PyFlasher - is there a recommended tool for flashing these boards?

kymodoke, 2021-11-10:
I don't use the official programmer but a simple CP2102 USB to UART bridge (as pins on the side for the programmer are just plain 3V-GND-.-.-RX-TX).
I always use the good old and reliable [esptool.py](https://github.com/espressif/esptool) to flash my ESP boards, and also for my LilyGO-T-ETH-POE boards.

Here how I do with my CP2102 programmer:
1. Remove any power source (like USB-C), and don't plug yet the USB cable of the programmer to your computer.
2. press the "BOT" button on the board and keep it pressed.
3. plug the USB cable of your programmer (it will power on the board), while still keeping the BOT button pressed.
4. release the button after 2 seconds.
-> the board will boot in programming mode.

vanroy-b, 2022-05-13:
We've been unable to flash the device when an sd card is inserted, we remove it to flash new firmware

closed 2023-02-12 by lewisxhe

OmegaModr, 2023-03-24:
Hi, can confirm that the firmware is working, mind including it in your regular builds?

I'll leave it running if it encounters any issues