Xinyuan-LilyGO/LilyGO-T-ETH-Series #17: PoE doesn't work

thiagohd, 2022-09-08:
I have powered with a 24V PoE adapter which I have from a router, but it didn't work.
Checking the temperature of the SI3404 controller, its temperature is around 70 Celsius degrees.

The voltage reaches C32, but VDD is always 0 V (VSS referenced).
I have checked D4 and C35, but they seem fine.

Did anyone have this problem too?

kimble4, 2022-09-08:
It's designed for proper 802.3af PoE, and from the datasheet, the SI3404 
regulator requires 37-57V to operate.

I'm successfully powering a couple of these using 'passive' injectors 
(ie. those which simply supply power at all times, rather than 
negotiating with the device before supplying power as per the 802.3af 
spec) with a 48V supply.

Kim.

thiagohd, 2022-09-09:
@kimble4 You are absolutely right. The regulator requires 37-57V to operate.

Thank you. Let's close this.

closed 2022-09-09 by thiagohd