Xinyuan-LilyGO/LilyGO-T-ETH-Series #18: Schematic PoE ( V+) Signal Goes Nowhere?

creepscompilation, 2022-09-12:
Hello,

The "SI304" IC on the schematic has a signal named (V+) that does not connect to the rest of the circuit?

So, based on my reading of the schematic, the "SI3404" IC provides the power from the PoE connection, yet
I don't see where the +5vdc power comes from when only PoE is the power source for the board?

Is this signal Mislabeled?
Please see the attached schematic and screenshot.

[T-ETH-POE.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/files/9549984/T-ETH-POE.pdf)
[image]

creepscompilation, 2022-09-12:
[Uploading T-ETH-POE.pdf…]()

kimble4, 2022-09-12:
It's on that fragment of circuit below the regulator block, connected to Q3 and D5.  (Yes, it took me ages to find, I hate schematics that over-use labelled signals like this.)

creepscompilation, 2022-09-12:
Thank you Kimble4..
I get it now...

For my project, when PoE power is not available, I want to use a standard 5vdc power adaptor as an alterative power source instead of USB.

Just to confirm, the (V+) signal is actually 5vdc?
I noticed the notation for VBUS is (3.7v to 4.2v)?

Basically, my project will need to the Ethernet connection to work whether PoE power is available or not.

Finally, I don't see on the schematic a USB serial IC that would normally be used to UPLOAD the Arduino code?
I find connector "P1" has the UART connections, am I supposed to have an external USB to serial adaptor?

closed 2023-02-12 by lewisxhe