Xinyuan-LilyGO/LilyGO-T-ETH-Series #21: Board won't power up anymore

fraglord, 2023-01-21:
The board (v1.1) was working fine for a few days on PoE. Today I moved the PoeSwitch and the board to it's final destination, plugged everythign together and - nothing! No lights come up or ethernet connection established. Same result with alternative power via USB-C. With the daughter-board I can still power it up and program the ESP32.
So something wrong with the supply. On the input pin of U4 I do not see any voltage. And on D5 ther is a flaky voltage of a few hundred mV. Unfortunately there is no parts print on the board and I am not that good at trouble shooting. Any help appreciated!

lewisxhe, 2023-02-09:
DCan you take a picture to indicate where it is?

lewisxhe, 2023-02-18:
No reply for a long time, please reopen it if you have any questions

closed 2023-02-18 by lewisxhe