Xinyuan-LilyGO/LilyGO-T-ETH-Series #23: Multiple SPI Interfaces

martinmmi, 2023-02-12:
Hello, i have a question. I am working with the T-ETH-POE Board and an SPI-LoRa-Shield and an 0,96 Monochrom SPI OLED.

For the OLED i used this lib: https://github.com/olikraus/u8g2
And for LORA i used this lib: https://github.com/sandeepmistry/arduino-LoRa

That is the constructor for the OLED:
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ DISPLAY_CS, /* dc=*/ DISPLAY_SCLK, /* reset=*/ DISPLAY_RST);

And here iam initializing the SPI for LORA:
SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI, LORA_CS);

When i am initializing the OLED and LORA as above in writing the code together, i get the error: SPI-Connection to LORA Modul failed.  Without  U8G2 it works fine. Everytime when i communicate with one SPI-Interface, i put the CS-Signal LOW.

How can i work with multiple spi interfaces with the LilyGO-T-ETH board?

Many thanks!

lewisxhe, 2023-02-12:
Which pins do you use?

martinmmi, 2023-02-12:
MOSI 15, MISO 02, SCLK 14 und for CS for each spi device on gpio.

lewisxhe, 2023-02-12:
When sharing SPI bus with SD card, you need to disable the CS of other SPI, that is, set it to HIGH

martinmmi, 2023-02-12:
Yeah, i did it. But i think there is somewhere a other error that i cant work with display and lora together. Maybe, i wait for the spi oled and when it connected, i will try again

lewisxhe, 2023-02-12:
33 None of the above can be used as output, that is, as CS

martinmmi, 2023-02-12:
Which pins i can use than for CS? Are every GPIO only inputs?

lewisxhe, 2023-02-12:
MISO INPUT
MOSI OUTPUT
SCK OUTPUT
CS OUTPUT

martinmmi, 2023-02-12:
Yeah, but what is in my case, when i connect a other spi device? What is with the gpios in the right upper corner? There are only input??? 

[image]

lewisxhe, 2023-02-12:
[image]

martinmmi, 2023-02-12:
Thanks so much. I will try it when i have the display again. 👍

lewisxhe, 2023-02-18:
Please reopen it if there is any problem

closed 2023-02-18 by lewisxhe