Xinyuan-LilyGO/LilyGO-T-ETH-Series #24: I2C on Downloaderboard available?

martinmmi, 2023-02-12:
Hello, is there a pindocumentation for the original LilyGO Downloaderboard available?

Thanks much
Martin

lewisxhe, 2023-02-12:
[image]
As can be seen from the schematic diagram, it only connects the pin for download

closed 2023-02-12 by lewisxhe

martinmmi, 2023-02-12:
Okay, and the GPIOs on the Downloader Board?

lewisxhe, 2023-02-12:
The only GPIO0 and RX TX, are you sure you want to use them?

martinmmi, 2023-02-12:
I mean this pins. IO12, 4, 15, 2, 13, 14, 0

But yeah, i see it now. Thats only io what you need for the download board. Only 0, RX and TX are new at this board.

[image]

martinmmi, 2023-02-12:
Why the I2C pins are not available from the esp32?

lewisxhe, 2023-02-12:
I understand what you mean. These are not available. For how to use other pins, please see here https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/examples/SPI_Wire_DevicesExample/SPI_Wire_DevicesExample.ino