Xinyuan-LilyGO/LilyGO-T-ETH-Series #27: SD-Card Issue

martinmmi, 2023-02-20:
Hey Lewis,

one question. When i try to execute the SharingSPIBus example, the sd-interface will never mounted. 

I still have my Lora Shield attached to the SPI-Bus (PIN 2, 14, 15) with a separate CS-PIN. Since this is not addressed by the example, I cannot imagine that the error came from this?

What can i do? 

[image]

lewisxhe, 2023-02-21:
I made a simple example of LoRa sending. You can refer to the following
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/tree/master/examples/LoRaShiled

martinmmi, 2023-02-21:
Hey Lewin, if I also correctly connect an SPI display to the SPI bus, the SD card reader fails every time during initialization (without having programmed the display). Is there anything else that can be done? Without a connected display, it works.

lewisxhe, 2023-02-22:
Please ensure that the CS of your other SPI devices is low. If you use the SPI display, please ensure that the screen RST is high and the CS is high, so that access to the sd card will not be occupied by the bus

martinmmi, 2023-02-22:
`
//////////////////////////////////////////////////////////////////////

void setup() {

    Serial.begin(115200);
    while (!Serial);

    Serial.println("");
    Serial.println(name);
    Serial.println("Version: " + version);

    pinMode(DISPLAY_CS, OUTPUT);                                //CS Display
    pinMode(LORA_CS, OUTPUT);                                   //CS LORA
    pinMode(SD_CS, OUTPUT);                                     //CS SD-Reader
    pinMode(SD_MISO, INPUT_PULLUP);

    digitalWrite(LORA_CS, HIGH);
    digitalWrite(DISPLAY_CS, HIGH);
    digitalWrite(SD_CS, LOW);

    pinMode(DISPLAY_RST, OUTPUT);                               //Not sure
    digitalWrite(DISPLAY_RST, HIGH);

//////////////////////////////////////////////////////////////////////

    SPI.begin(SD_SCLK, SD_MISO, SD_MOSI);                      

    while (true) {
            if (SD.begin(SD_CS)) {
                Serial.println("SDCard init succeeded.");
                delay(300);
                break;
            }
            Serial.println("SDCard init failed. Check your connections.");
            delay(300);
            break;
        }

    digitalWrite(LORA_CS, HIGH);
    digitalWrite(DISPLAY_CS, LOW);
    digitalWrite(SD_CS, HIGH);         //Select Display SPI Device

//////////////////////////////////////////////////////////////////////

    u8g2.begin();
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setContrast(defaultBrightnessDisplay);                  
    u8g2.setFlipMode(1);

    //        Color, Delay, Runs
    printLogo(0, 50);
    delay(1000);
    printLoad(1, 60, 2);
    printLogo(0, 25);
    delay(500);
    printLoad(1, 60, 4);

    sprintf(buf_version, "%s", version);
    u8g2.drawStr(99,60,buf_version);
    u8g2.sendBuffer();

    Serial.println("OLED init succeeded.");
    oledInit = "OLED init";
    sprintf(buf_oledInit, "%s", oledInit);
    u8g2.drawStr(0,15,buf_oledInit);
    u8g2.sendBuffer();
    delay(300);

    digitalWrite(LORA_CS, LOW);
    digitalWrite(DISPLAY_CS, HIGH);
    digitalWrite(SD_CS, HIGH);                                        //Select Display SPI Device 

//////////////////////////////////////////////////////////////////////  

    LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);
    LoRa.setTxPower(loraTxPower);
    LoRa.setSpreadingFactor(loraSpreadingFactor);    
    LoRa.setSignalBandwidth(loraSignalBandwidth);
    LoRa.setCodingRate4(loraCodingRate);
    LoRa.setPreambleLength(loraPreambleLength);
    LoRa.begin(loraFrequenz);

    if (!LoRa.begin(loraFrequenz)) {                                 // initialize lora frequenz
        Serial.println("LORA init failed. Check your connections.");
        loraInit = "LORA failed";
        digitalWrite(LORA_CS, HIGH);
        digitalWrite(DISPLAY_CS, LOW);
        digitalWrite(SD_CS, HIGH);                                  //Select Display SPI Device    
        sprintf(buf_loraInit, "%s", loraInit);   
        u8g2.drawStr(0,35,buf_loraInit);
        u8g2.sendBuffer();
        digitalWrite(DISPLAY_CS, HIGH);     
        while (true);                                               // if failed, do nothing
    }

    digitalWrite(LORA_CS, HIGH);
    digitalWrite(DISPLAY_CS, LOW);
    digitalWrite(SD_CS, HIGH);                                       //Select Display SPI Device   

//////////////////////////////////////////////////////////////////////

    Serial.println("LORA init succeeded.");
    loraInit = "LORA init";
    sprintf(buf_loraInit, "%s", loraInit);   
    u8g2.drawStr(0,35,buf_loraInit);
    u8g2.sendBuffer();
    delay(300);

    pinMode(gpioP1, INPUT_PULLDOWN);
    pinMode(gpioP2, INPUT_PULLDOWN);
    pinMode(gpioP3, INPUT_PULLDOWN);
    pinMode(gpioP4, INPUT_PULLDOWN);

    Serial.println("Outputs init succeeded.");
    outputInit = "Outputs init";
    sprintf(buf_outputInit, "%s", outputInit);   
    u8g2.drawStr(0,45,buf_outputInit);
    u8g2.sendBuffer();
    delay(500);

    printLora(1);
    delay(2500);

    emptyDisplay();
    printDisplay();

    digitalWrite(LORA_CS, HIGH);
    digitalWrite(DISPLAY_CS, HIGH);
    digitalWrite(SD_CS, HIGH);         //Select Display SPI Device   

//////////////////////////////////////////////////////////////////////

    if(!SPIFFS.begin(true)){
        Serial.println("SPIFFS init failed. An Error has occurred while mounting SPIFFS");
        return;
    }
    Serial.println("SPIFFS init succeeded.");

//////////////////////////////////////////////////////////////////////

    WiFi.onEvent(WiFiEvent);

    pinMode(NRST, OUTPUT);

    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);
    delay(200);
    digitalWrite(NRST, 0);
    delay(200);
    digitalWrite(NRST, 1);

    ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
`

martinmmi, 2023-02-22:
That is my setup routine. I could understand it. Everytime, when iam working with the spi bus, i set the spi cs to low and the other to high. After working, i set the other spi cs low and the others high. :(

lewisxhe, 2023-02-22:
If you use the library, the<CS>has been set to Low in the library. You only need to ensure that the required device is selected for initialization when initializing. The following operation library has already operated on the<CS>

martinmmi, 2023-02-25:
Hey lewis, today i found out, when iam setting (after access to spi) the SPI.end(); and start SPI.begin(...) again, then it works fine. 

I couldnt understand why the spi didnt work. Can you try again to run your lora board with a spi oled?

I wrote the program for this:

//////////////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////////

#include <Arduino.h>
#include <U8g2lib.h>
#include <LoRa.h>
#include <Preferences.h>                //lib for flashstoreage
#include <bitmaps.h>
#include <ETH.h>
#include <SPI.h>
#include <SD.h>
#include <SPIFFS.h>                     //lib for filesystem
#include <ESPAsyncWebServer.h>
#include <ESPmDNS.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

int counter = 0;

String mode = "discover";

String outgoing;

char buf_tx[12];

///////////////////////////////////////////////
///////// Setup Transmitter Values ////////////
///////////////////////////////////////////////

byte localAddress = 0xaa;                 // Address of this device  
String string_localAddress = "aa";                                    
byte destination = 0xff;                  // Destination to send to              
String string_destinationAddress = "ff";            

///////////////////////////////////////////////
///////////////////////////////////////////////
///////////////////////////////////////////////

byte msgKey1 = 0x2a;                      // Key of outgoing messages
byte msgKey2 = 0x56;
byte msgCount = 0;                        // Count of outgoing messages
byte esm = 0x00;                          // 0x00 -> OFF  0x01 -> ON

int defaultBrightnessDisplay = 150;   // value from 1 to 255

///////////////////////////////////////////////
//////////// Setup LORA Values ////////////////
///////////////////////////////////////////////

int loraTxPower = 17;                   //2-20 default 17
int loraSpreadingFactor = 7;            //6-12 default  7
double loraSignalBandwidth = 125E3;     //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, 250E3, and 500E3 default 125E3
int loraCodingRate = 5;                 //5-8 default 5
int loraPreambleLength = 8;             //6-65535 default 8
double loraFrequenz = 868E6;            //set Frequenz 915E6 or 868E6

///////////////////////////////////////////////
[... 94 lines cut ...]
    sendMessage(outgoing);
    //digitalWrite(LORA_CS, HIGH);
    SPI.end();

    SPI.begin(DISPLAY_SCLK, DISPLAY_MISO, DISPLAY_MOSI);           
    //digitalWrite(DISPLAY_CS, LOW);
    u8g2.begin();
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setContrast(defaultBrightnessDisplay);                  
    u8g2.setFlipMode(0);
    u8g2.drawStr(0,45,buf_tx);
    u8g2.sendBuffer();
    //digitalWrite(DISPLAY_CS, HIGH);
    SPI.end();

    Serial.println("LORA TxD: " + outgoing);

    delay(2000);

}

//////////////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////////
//////////////////////////////////////////////////////////////////////

lewisxhe, 2023-02-27:
Change it to this way. You passed in the wrong DC Pin in the OLED constructor. The parameter you passed in is MISO

```
#define LORA_MISO                            2  //Pinout Section
#define LORA_MOSI                           15
#define LORA_SCLK                           14
#define LORA_CS                             33
#define LORA_RST                            12
#define LORA_IRQ                             4  //Must be a Hardware Interrupt Pin
#define DISPLAY_CS                          16
#define DISPLAY_RST                         32

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ DISPLAY_CS, /* dc=*/ -1, /* reset=*/ DISPLAY_RST);

//////////////////////////////////////////////////////////////////////

void sendMessage(String message)
{
    LoRa.beginPacket();                   // start packet
    LoRa.write(destination);              // add destination address
    LoRa.write(localAddress);             // add sender address
    LoRa.write(msgKey1);                  // add message KEY
    LoRa.write(msgKey2);                  // add message KEY
    LoRa.write(esm);                      // add energy save mode
    LoRa.write(msgCount);                 // add message ID
    LoRa.write(message.length());         // add payload length
    LoRa.print(message);                  // add payload
    LoRa.endPacket();                     // finish packet and send it
    msgCount++;                           // increment message ID
}

//////////////////////////////////////////////////////////////////////

void setup()
{

    Serial.begin(115200);

    pinMode(LORA_CS, OUTPUT);                                   //CS LORA
    pinMode(DISPLAY_CS, OUTPUT);                                //CS Display

    digitalWrite(LORA_CS, HIGH);
    digitalWrite(DISPLAY_CS, HIGH);

    SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI);
    LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ);
    LoRa.setTxPower(loraTxPower);
    LoRa.setSpreadingFactor(loraSpreadingFactor);
    LoRa.setSignalBandwidth(loraSignalBandwidth);
    LoRa.setCodingRate4(loraCodingRate);
    LoRa.setPreambleLength(loraPreambleLength);
    LoRa.begin(loraFrequenz);

    if (!LoRa.begin(loraFrequenz)) {                                        // initialize lora frequenz
        Serial.println("LORA init failed. Check your connections.");
        while (true);                                                       // if failed, do nothing
    }
    Serial.println("LORA init succeeded.");

    u8g2.begin();
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setContrast(defaultBrightnessDisplay);
    u8g2.setFlipMode(0);

}

//////////////////////////////////////////////////////////////////////

void loop()
{

    destination = 0xff;
    string_destinationAddress = "ff";
    outgoing = "dis-anyrec?";         // Send a message
    counter++;
    sprintf(buf_tx, "%d", counter);

    sendMessage(outgoing);

    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setContrast(defaultBrightnessDisplay);
    u8g2.setFlipMode(0);
    u8g2.drawStr(0, 45, buf_tx);
    u8g2.sendBuffer();
    Serial.println("LORA TxD: " + outgoing);
    delay(2000);

}

```