Xinyuan-LilyGO/LilyGO-T-ETH-Series #29: How do I place the board in "Flash Mode"

cheeseshark, 2023-03-05:
Normally, to place an ESP32 into "Flash Mode" to allow for the upload of a program, you have to press the buttons.

With there being 2 sets of buttons (one set on the Downloader module, and another set on the board itself), it is unclear on how to place the board into a mode that I can upload a program.

I am attempting to upload using the esptool software.

Thanks for the assistance.

-- Cheeseshark

lewisxhe, 2023-03-08:
If you use the downloader of LILYGO, you only need to connect the downloader correctly and then click Download to use it. If you can't download normally, please press and hold the BOOT button, don't let go, and then press RST. At this time, the processor is in download mode. At this time, you can download the program to the processor. After the download is completed, you need to press RST to let the processor exit the download mode.

cheeseshark, 2023-03-08:
I got it working, thank you.