Xinyuan-LilyGO/LilyGO-T-ETH-Series #38: Burned out PoE power

artbrayko, 2023-06-30:
[image]
[image]

Hello, PoE power burned out. Please tell me, can you give a schematic diagram of the device in order to determine the burnt elements? The microcircuit marked with a red arrow is very hot. It burned down on three devices of the same revision and the same production date. At the same time, the device is powered by USB Type-C + data over Ethernet. Can your technical engineer tell me what elements need to be replaced? Thank you.

GalaxyGateway, 2023-06-30:
I've had the same issue when back feeding through the 3.3V while connected to POE. The POE part burned up, the ethernet side itself continues to work. If someone could shed some light on this it would be great :-)

quakec, 2023-06-30:
Can you both clarify how long each board ran before the chip burned up?
I am currently in the process of testing mine long-term on PoE.

artbrayko, 2023-06-30:
A lot of these boards have gone through me. With revision v1.1 there were never any problems. These with revision v1.0 each worked for about a month. I want to fix them, replace burnt parts.

quakec, 2023-06-30:
That is good to know. There is only this schematic on this repository.

[https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/schematic/T-ETH-POE.pdf](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/schematic/T-ETH-POE.pdf)

artbrayko, 2023-06-30:
> 

Thanks, I'll take a look. But maybe someone from the manufacturer will comment on this?

closed 2024-03-28 by github-actions[bot]