Xinyuan-LilyGO/LilyGO-T-ETH-Series #41: SYSTEM_EVENT_ETH_CONNECTED never called

sergeybrazhnik-cb, 2023-07-20:
I have board revision 1.1.
Using regular usb-type-c as power and LAN cable as connection to my local network 

When Im running [example](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/tree/master/examples/UDPClientSender) I see 
ARDUINO_EVENT_ETH_START
ARDUINO_EVENT_ETH_CONNECTED
but **SYSTEM_EVENT_ETH_CONNECTED** never called! 
What could be the issue?

kimble4, 2023-07-20:
The events got renamed at some point.
SYSTEM_EVENT_ETH_CONNECTED is now ARDUINO_EVENT_ETH_CONNECTED

quakec, 2023-07-20:
ARDUINO_EVENT_ETH_CONNECTED  == SYSTEM_EVENT_ETH_CONNECTED 

They are the same.  I had to change mine from ARDUINO_ to SYSTEM_ 

Depending on your IDE you'll use either one, the values are the same.

sergeybrazhnik-cb, 2023-07-24:
@kimble4 @quakec 
Thanks for the response guys! 
This events are coming and this is fine. But the **SYSTEM_EVENT_ETH_CONNECTED**  is missing. That's why board did not obtains the IP address and not able to communicate in LAN

sergeybrazhnik-cb, 2023-07-25:
OMG, I see that I made a really mess in the events names.
One more time
I see that 
ARDUINO_EVENT_ETH_START
ARDUINO_EVENT_ETH_CONNECTED
Called successfully but 
**ARDUINO_EVENT_ETH_GOT_IP** never get called
What could be the reason? 

BTW do you know how to enable the verbose debugging mode to see potential places where it went wrong?

quakec, 2023-07-25:
> OMG, I see that I made a really mess in the events names. One more time I see that ARDUINO_EVENT_ETH_START ARDUINO_EVENT_ETH_CONNECTED Called successfully but **ARDUINO_EVENT_ETH_GOT_IP** never get [...]

Got ARDUINO_EVENT_ETH_GOT_IP won't be switched to in the event method if you have set a static IP address.

sergeybrazhnik-cb, 2023-07-25:
Im using [this](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/examples/UDPClientSender/UDPClientSender.ino) example to test. There is no static IP set. 

My local network is 1 PC on linux, then switch with poe support then LILYGO device.

quakec, 2023-07-25:
> Im using [this](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE/blob/master/examples/UDPClientSender/UDPClientSender.ino) example to test. There is no static IP set.
> 
> My local network is 1 PC on linux, then switch with poe support then LILYGO device.

It sounds like a DHCP server can't be reached from the your LILYGO device.

sergeybrazhnik-cb, 2023-08-08:
@quakec you were right! 
Thanks for the advice! 

Actually I have used small network with 1 linux machine and 1 lilygo connected over switch where were no DHCP server.
After switching to other network LILYGO started obtaining IP address