Xinyuan-LilyGO/LilyGO-T-ETH-Series #43: Where to buy the T-ETH-LITE-ESP32S3

jonasniesner, 2023-08-02:
I have not found this board on the website yet. This board looks like a perfect fit for a project of mine. Is there a timeline when it and the Poe add-on board will be available (preferably on AliExpress)?

quakec, 2023-08-02:
> I have not found this board on the website yet. This board looks like a perfect fit for a project of mine. Is there a timeline when it and the Poe add-on board will be available (preferably on AliExpress)?

Are you referring to this board:

Just found this amazing item on AliExpress. Check it out! ￡5.73 | LILYGO® TTGO T-Lite W5500 ESP32 Master Chip SSD1306 0.96-inch OLED Type-C USB Programming Board WiFi Bluetooth Wireless Module
https://a.aliexpress.com/_mO1D9Py

lewisxhe, 2023-08-02:
There is currently no sales going on, we need to wait for a period of time

jonasniesner, 2023-08-02:
Ok, is there a way to get informed when the board is available for sale? Is the design of the board final, so I can start the design of a hat for it?

lewisxhe, 2023-08-02:
If it is sold, I will notify you here. Regarding other design resources, I have made it public. You can view the structure size in the shell directory, and the schematic directory has schematic diagrams

lewisxhe, 2023-09-16:
Currently available for purchase, please check https://www.lilygo.cc/products/t-eth-lite?variant=43120880779445

closed 2023-09-16 by lewisxhe