Xinyuan-LilyGO/LilyGO-T-ETH-Series #44: Can't connect using a static IP

rapdSagarbe, 2023-08-14:
use that code

    // Use static ip address config
    IPAddress local_ip(192, 168, 1, 44);
    IPAddress gateway(192, 168, 1, 1);
    IPAddress subnet(255, 255, 255, 0);
    IPAddress dns(8, 8, 8, 8);
    IPAddress dns2(8, 8, 4, 4);

    ETH.config(local_ip,
              gateway,
              subnet,
              dns,
              dns2);

AND when try to conect a server using this code:

void sendPOST(String data) {
  HTTPClient http;

  http.begin("http://MY_SERVER/test.php");

  http.addHeader("Content-Type", "application/x-www-form-urlencoded");
  http.addHeader("Content-Length", String(data.length()));

  int httpResponseCode = http.POST(data);

Serial.println(httpResponseCode);

  // Comprobar el código de respuesta
  if (httpResponseCode == 200) {
    String response = http.getString();
    Serial.println(response);
  } else if(httpResponseCode >= 400 and httpResponseCode<500)
  {
    Serial.println("No encontrado");
  } else if(httpResponseCode >=500)
  {
    Serial.println("Error del servidor");
  }
  else {
    // Imprimir el mensaje de error
    Serial.print("Error al enviar la solicitud POST: ");
    Serial.println(http.errorToString(httpResponseCode).c_str());
  }

  // Cerrar la conexión
  http.end();
}

RECIBE this msg: 

11:09:38.941 -> -1
11:09:38.941 -> Error al enviar la solicitud POST: connection refused

WHEN use dinamic ip, don't have any problem to connect.

Thank for all help.

0x3333, 2023-08-15:
Before `ETH.config` run `ETH.begin()`.

closed 2023-08-16 by lewisxhe

rapdSagarbe, 2023-08-16:
no connect propertly, using exaple code

/**
 * @file      StaticIPAddress.ino
 * @author    Lewis He (lewishe@outlook.com)
 * @license   MIT
 * @copyright Copyright (c) 2023  Shenzhen Xin Yuan Electronic Technology Co., Ltd
 * @date      2023-02-17
 *
 */
#include <Arduino.h>
// #include <ETH.h>
#include <ETHClass.h>       //Is to use the modified ETHClass
#include <SPI.h>
#include <SD.h>
#include "utilities.h"          //Board PinMap

//Change to IP and DNS corresponding to your network, gateway
IPAddress staticIP(192, 168, 1, 37);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(192, 168, 1, 1);

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
    switch (event) {
    case ARDUINO_EVENT_ETH_START:
        Serial.println("ETH Started");
        //set eth hostname here
        ETH.setHostname("esp32-ethernet");
        break;
    case ARDUINO_EVENT_ETH_CONNECTED:
        Serial.println("ETH Connected");
        break;
    case ARDUINO_EVENT_ETH_GOT_IP:
        Serial.print("ETH MAC: ");
        Serial.print(ETH.macAddress());
        Serial.print(", IPv4: ");
        Serial.print(ETH.localIP());
        if (ETH.fullDuplex()) {
            Serial.print(", FULL_DUPLEX");
        }
        Serial.print(", ");
        Serial.print(ETH.linkSpeed());
        Serial.println("Mbps");
        eth_connected = true;
        break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
        Serial.println("ETH Disconnected");
        eth_connected = false;
        break;
    case ARDUINO_EVENT_ETH_STOP:
        Serial.println("ETH Stopped");
        eth_connected = false;
        break;
    default:
        break;
    }
}

void testClient(const char *host, uint16_t port)
{
    Serial.print("\nconnecting to ");
    Serial.println(host);

    WiFiClient client;
    if (!client.connect(host, port)) {
        Serial.println("connection failed");
        return;
    }
    client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
    while (client.connected() && !client.available())
        ;
    while (client.available()) {
        Serial.write(client.read());
    }

    Serial.println("closing connection\n");
    client.stop();
}

void setup()
{
    Serial.begin(115200);

    WiFi.onEvent(WiFiEvent);

#ifdef ETH_POWER_PIN
    pinMode(ETH_POWER_PIN, OUTPUT);
    digitalWrite(ETH_POWER_PIN, HIGH);
#endif

    if (ETH.config(staticIP, gateway, subnet, dns, dns) == false) {
        Serial.println("Configuration failed.");
    }

#if CONFIG_IDF_TARGET_ESP32
    if (!ETH.begin(ETH_ADDR, ETH_RESET_PIN, ETH_MDC_PIN,
                   ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE)) {
        Serial.println("ETH start Failed!");
    }
#else
    if (!ETH.beginSPI(ETH_MISO_PIN, ETH_MOSI_PIN, ETH_SCLK_PIN, ETH_CS_PIN, ETH_RST_PIN, ETH_INT_PIN)) {
        Serial.println("ETH start Failed!");
    }
#endif

while (!eth_connected) {
  Serial.print(".");
  delay(10);
}

Serial.println("IP:");
Serial.println(ETH.localIP().toString());
}

void loop()
{
    if (eth_connected) {
        testClient("www.baidu.com", 80);
    }
    Serial.println("IP:");
Serial.println(ETH.localIP().toString());

    delay(10000);
}

lewisxhe, 2023-08-16:
Tey use ETH.h
#include <ETH.h>
//#include <ETHClass.h> //Is to use the modified ETHClass

0x3333, 2023-08-16:
Take a look at this example: [ETH_LAN8720.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino) Just add the static IP configuration before begin, good to go.

When you copy/paste code, use the "Code" function in the toolbar, it's impossible to read.

kilrah, 2023-09-23:
> Before `ETH.config` run `ETH.begin()`.

This works, but then the example in this repo needs to be fixed since it specifically does the config() before the begin()
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/examples/HelloServer/HelloServer.ino#L125