Xinyuan-LilyGO/LilyGO-T-ETH-Series #45: T-ETH-Lite-ESP32S3: Unable to verify flash chip connection

MattGrayYes, 2023-11-04:
Hiya

Apologies if I'm doing something silly, but I havent been able to upload exmaple code to the T-ETH-Lite-ESP32S3 boaerd.

Something's working because It's able to get the board info.

Any suggestions on what I can do next please? Details below.

Thanks!
Matt

```
Sketch uses 810413 bytes (25%) of program storage space. Maximum is 3145728 bytes.
Global variables use 47196 bytes (14%) of dynamic memory, leaving 280484 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-210
Connecting....
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 84:fc:e6:6c:cc:10
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
Failed uploading: uploading error: exit status 2
```

To reproduce:

1. Open HelloServer.ino example code in arduino
2. #define LILYGO_T_ETH_LITE_ESP32S3 in utilities.h
3. Set board settings as per instructions (screenshot below)
4. Plug USB TTL cable in
5. Press and hold boot, press reset, let go of boot.
6. Click upload in Arduino
7. Compiles successfully
8. Above output appears.

[image]

lewisxhe, 2023-11-04:
Hi  

Very detailed instructions. You can use the following steps to confirm whether it has entered download mode?

1. Connect the external USB TO TTL and connect it correctly. Please check the README for instructions on how to connect.
2. Open the Serial monitor
3. Press RST and check whether there is startup information. If there is no startup information, the USB TO TTL connection may be incorrect.
4. Press and hold the BOOT button, don't release it, then click the RST button, release the BOOT button, and check whether the Serial outputs the words "Waiting for download".
If the word "Waiting for download" appears in the fourth step, try uploading the sketch again. If it fails, please try lowering the baud rate.

MattGrayYes, 2023-11-06:
Thanks for the help.
I bought a new cable just in case and it worked immediately.

### For anyone else reading this having had a similar issue
I'm now using the official [FTDI TTL-232R-3V3](https://ftdichip.com/products/ttl-232r-3v3/) cable, 
- [FTDI TTL-232R-3V3 Datasheet](https://ftdichip.com/wp-content/uploads/2023/07/DS_TTL-232R_CABLES.pdf)
- Buy: [FTDI TTL-232R-3V3 at Digikey](https://www.digikey.co.uk/en/products/detail/ftdi-future-technology-devices-international-ltd/TTL-232R-3V3/1836393)

With 4 breadboard / dupont connector pin to socket leads
- [Similar product at Digikey](https://www.digikey.co.uk/en/products/detail/mikroelektronika/MIKROE-2314/8564418)

closed 2023-11-06 by MattGrayYes

tiagoad, 2024-01-27:
In my case (LilyGO T3 v1.6), I needed to change Upload Speed to 115200 in the Tools menu.

Zkall123, 2024-04-15:
my board dont go in Downloadmode if i press the boot boot then the RST butten and release the RST button