Xinyuan-LilyGO/LilyGO-T-ETH-Series #47: StaticIPAddress example doesn't work

gunicsba, 2023-11-25:
Hello,

Board used: ESP32-S3 T-ETH-Lite and WT5500

According to issue #44 we'd have to call Config AFTER ETH.begin();

I believe this line is way too early in the code:
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/935d50d529eb077baae9aaa81f534e22e0df95aa/examples/StaticIPAddress/StaticIPAddress.ino#L96-L98

If I move Config after the beginSPI() it works as expected.

If I do the Config in the WiFi OnEvent  ARDUINO_EVENT_ETH_START  or ETH_CONNECTED then it works as well.

gunicsba, 2023-11-25:
This is my proposed fix:
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/compare/master...gunicsba:LilyGO-T-ETH-Series:staticipfix?expand=1

lewisxhe, 2023-11-26:
I only wrote the static IP and did not test it. Thank you for your feedback and updates.

lewisxhe, 2023-11-26:
Thank you for your feedback and updates. Could you please submit a PR

lewisxhe, 2023-11-26:
Already merged

closed 2023-11-26 by lewisxhe