Xinyuan-LilyGO/LilyGO-T-ETH-Series #49: T-Eth-Lite S3 W5500 crashes on AsyncUDP

troyhacks, 2023-12-10:
Example code here adapted from the WiFiUDP example:

https://github.com/troyhacks/LilyGO-T-ETH-Series/blob/master/examples/UDPClientReceiver/UDPClientReceiver.ino

Works great with WiFiUDP, crashes without error or a meaningful backtrace with AsyncUDP.

I've tried to do some debugging but I'm posting here in case there's any insights. 

I would love to get this board running with AsyncUDP as I'm working towards adding support for it to [WLED](https://github.com/MoonModules/WLED) and this is a blocking issue from finishing the integration.

Thanks in advance for any hints!

```
ETH Started
ETH Connected
ETH MAC: 34:85:18:8F:40:25, IPv4: 192.168.8.44, FULL_DUPLEX, 100Mbps
Starting UDP listener on port 6454
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4005544b  PS      : 0x00060730  A0      : 0x8201760b  A1      : 0x3fcf6fc0
A2      : 0x00000000  A3      : 0x3fceee00  A4      : 0x803803b9  A5      : 0x3fced490
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x82017440  A9      : 0x3fcf6f70  
A10     : 0x00000001  A11     : 0x3fcf199c  A12     : 0x3fcf1950  A13     : 0x00000000
A14     : 0x3fcf199c  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40055499  LEND    : 0x400554a9  LCOUNT  : 0xfffffffb

Backtrace: 0x40055448:0x3fcf6fc0 |<-CORRUPTED

  #0  0x40055448:0x3fcf6fc0 in ?? ??:0
```

lewisxhe, 2023-12-18:
I have not tested AsyncUDP. Looking at the error report, it should be that the null pointer was accessed and caused the program to crash.

lewisxhe, 2023-12-18:
I have found the cause of the crash, please disable it with the following lines and the crash will be resolved.
I can't continue to do other things. I have other things to deal with. I hope you can understand. You can report this error to espressif

[image]

troyhacks, 2023-12-19:
Much appreciated. Thank you!

closed 2023-12-19 by troyhacks