Xinyuan-LilyGO/LilyGO-T-ETH-Series #51: About DHCP Lease

AviStudio, 2024-01-02:
Hi! If I use DHCP IP address, will it automatically renew when lease time expires? If no, is there a solution for that? maybe to call a function to do that? Or the only is to use a Static IP ?

lewisxhe, 2024-01-02:
DHCP IP allocation and management is not managed by esp, but by the upper-level router.

AviStudio, 2024-01-02:
Yes, I understand, but the router gives an IP address for a specific time, what will happen when that time expires? Will ESP ask for a lease renewal ? or maybe there is a function like Ethernet.maintain() in Arduino "Ethernet" Library?

lewisxhe, 2024-01-02:
This behavior is determined by the router. If the IP expires and the DHCP server is forcibly disconnected, the ESP will disconnect and wait for network communication again. After the DHCP server completes the IP update, the ESP will obtain the address from the upper-level DHCP server again.

AviStudio, 2024-01-02:
I will check this and come with a feedback, Thanks!

AviStudio, 2024-03-13:
So, after 2 month of 24/7 connection time, I can confirm, that DHCP lease renewal is working without problems.

closed 2024-04-29 by github-actions[bot]