Xinyuan-LilyGO/LilyGO-T-ETH-Series #53: MDNS not working with T-ETH-Lite-ESP32S3

mathieucarbou, 2024-02-06:
I noticed that in my application: MDNS info is not published when only using ETH, is published when using both ETH and WiFi STA mode, but published with the ETH ip address.

I am running the `ETHOTA` example to reproduce,

[image]

I should see `esp32-ethernet.local`

Note: I even added the following juste before begin() to make sure ArduinoOTA published MDNS info.

```
ArduinoOTA.setHostname("esp32-ethernet");
ArduinoOTA.setMdnsEnabled(true);
```

lewisxhe, 2024-03-13:
oh ! I'm not sure about this issue at the moment. I need time to verify it.

mathieucarbou, 2024-04-14:
Still an issue

lewisxhe, 2024-04-18:
This function is already normal when tested in arduino 3.0.0-rc1 version, but there is still a problem in arduino 2.0.15.

lewisxhe, 2024-04-18:
There is no need to worry about this issue, because the future will definitely be based on v3.x, and I have added compatibility with v3.x.

mathieucarbou, 2024-04-21:
> There is no need to worry about this issue, because the future will definitely be based on v3.x, and I have added compatibility with v3.x.

I disagree... If I follow what you say, it means that MDNS won't be supported for all Lilygo ETH ESP32 S3 boards on Arduino 2. It will only work on Arduino 3.

What would be the solution to add MDNS support to the LIlygo T-ETH S3 board on Arduino 2 ? Is it possible ?

closed 2024-06-05 by github-actions[bot]