Xinyuan-LilyGO/LilyGO-T-ETH-Series #54: Can't compile any sketch for T-ETH-Lite-ESP32

PureTek-Innovations, 2024-03-02:
I'm using platformio, it is installed properly as I use it all the time for ESP32s.

I just want to get an ethernet connection up and running, so am trying to use the ETHOTA example.

I've tried using the default platform.ini with the correct board and sketch uncommented but I get this error:
`cc1plus.exe: fatal error: examples/ETHOTA/ETHOTA.ino.cpp: No such file or directory`

I've tried building my own sketch based on the example but the I get a bunch of errors from the ETHClass.cpp file:

```
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp: In member function 'bool ETHClass::beginSPI(int, int, int, int, int, int, spi_host_device_t, uint8_t, int, bool)':
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:314:5: error: 'eth_w5500_config_t' was not declared in this scope
     eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
     ^~~~~~~~~~~~~~~~~~
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:314:5: note: suggested alternative: 'eth_phy_config_t'
     eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
     ^~~~~~~~~~~~~~~~~~
     eth_phy_config_t
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:317:5: error: 'w5500_config' was not declared in this scope
     w5500_config.int_gpio_num = irq;
     ^~~~~~~~~~~~
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:317:5: note: suggested alternative: 'gpio_config'
     w5500_config.int_gpio_num = irq;
     ^~~~~~~~~~~~
     gpio_config
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:321:15: error: 'esp_eth_mac_new_w5500' was not declared in this scope
     eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
               ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:321:15: note: suggested alternative: 'esp_eth_mac_new_esp32'
     eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
               ^~~~~~~~~~~~~~~~~~~~~
               esp_eth_mac_new_esp32
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:327:15: error: 'esp_eth_phy_new_w5500' was not declared in this scope
     eth_phy = esp_eth_phy_new_w5500(&phy_config);
               ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/T-ETH-Lite-ESP32/ETHClass/src/ETHClass.cpp:327:15: note: suggested alternative: 'esp_eth_phy_new_ip101'
     eth_phy = esp_eth_phy_new_w5500(&phy_config);
               ^~~~~~~~~~~~~~~~~~~~~
               esp_eth_phy_new_ip101
*** [.pio\build\T-ETH-Lite-ESP32\libecf\ETHClass\ETHClass.cpp.o] Error 1
```

These are quite strange as I have declared the correct ethernet type and it is not W5500:
`#define ETH_TYPE                        ETH_PHY_RTL8201`

Are you able to provide a basic working platformio sketch and ini file, please?

PureTek-Innovations, 2024-03-02:
Please can you also explain why line 115 of the default platformio.ini has been included?

; ESP32-WROVER-E + RTL8201 FLASH:16MB PSRAM:8MB
[env:T-ETH-Lite-ESP32] 
board = esp32dev 
build_flags =  
	${env.build_flags} 
	-DBOARD_HAS_PSRAM 
    -DLILYGO_T_ETH_LITE_ESP32 
    -DUSER_SETUP_LOADED 
    -include lib/TFT_eSPI/User_Setups/Setup216_LilyGo_ETH_Lite_ESP32.h

This board does not have a TFT screen, does it need it for the SPI settings at the bottom of Setup216_LilyGo_ETH_Lite_ESP32.h?

PureTek-Innovations, 2024-03-12:
Are you using a different version of ETHClass.cpp?

Please try compiling with a  fresh unmodified download from github.

lewisxhe, 2024-03-12:
[image]
Try changing to the fixed version here

PureTek-Innovations, 2024-03-12:
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting ETHOTA.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ETHClass @ 1.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\T-ETH-Lite-ESP32\src\ETHOTA.ino.cpp.o
Building .pio\build\T-ETH-Lite-ESP32\bootloader.bin
Generating partitions .pio\build\T-ETH-Lite-ESP32\partitions.bin
**xtensa-esp32-elf-g++: error: examples/ETHOTA/ETHOTA.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files**
compilation terminated.
Compiling .pio\build\T-ETH-Lite-ESP32\lib61d\Update\HttpsOTAUpdate.cpp.o
Compiling .pio\build\T-ETH-Lite-ESP32\lib61d\Update\Updater.cpp.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\T-ETH-Lite-ESP32\libd0c\WiFi\WiFi.cpp.o
Compiling .pio\build\T-ETH-Lite-ESP32\libd0c\WiFi\WiFiAP.cpp.o
*** [.pio\build\T-ETH-Lite-ESP32\src\ETHOTA.ino.cpp.o] Error 1
Compiling .pio\build\T-ETH-Lite-ESP32\libd0c\WiFi\WiFiClient.cpp.o
====================================================== [FAILED] Took 138.96 seconds

PureTek-Innovations, 2024-03-12:
That didn't work

PureTek-Innovations, 2024-03-12:
> T-ETH-Lite-ESP32 supports TFT screen extension. The TFT_ILI9341_Shield example requires it. If you don’t use the screen, you can delete it. I will provide it to those who need it.

I can't find a TFT screen extension in the Lilygo shop, please can you let me have a link?

lewisxhe, 2024-03-12:
Can you provide AnyDesk for me to check?

lewisxhe, 2024-03-12:
> > T-ETH-Lite-ESP32 supports TFT screen extension. The TFT_ILI9341_Shield example requires it. If you don’t use the screen, you can delete it. I will provide it to those who need it.
> 
> I can't find a TFT screen extension in the Lilygo shop, please can you let me have a link?

https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/dda7a2ad4ab33d550c8dbaff5db1e61a0eda5aad/examples/TFT_ILI9341_Shield/TFT_ILI9341_Shield.ino#L10

PureTek-Innovations, 2024-03-12:
> Can you provide AnyDesk for me to check?

NO

PureTek-Innovations, 2024-03-12:
> https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/dda7a2ad4ab33d550c8dbaff5db1e61a0eda5aad/examples/TFT_ILI9341_Shield/TFT_ILI9341_Shield.ino#L10

That's a code example, not a product.

lewisxhe, 2024-03-12:
Then I have no choice. I suspect there is a problem with your platformio. I am creating a new github action to detect it.

PureTek-Innovations, 2024-03-12:
> Then I have no choice. I suspect there is a problem with your platformio. I am creating a new github action to detect it.

Or, more likely, a problem with your code.

PureTek-Innovations, 2024-03-12:
In fact, I fixed this days ago, I just wondered if the development team had any idea what they were doing.  It seems not.

Simply comment out line 15 in the example
//#include <ETHClass.h>       //Is to use the modified ETHClass

And uncomment line 14
#include <ETH.h>

Then the code compiles and works properly.

closed 2024-03-12 by PureTek-Innovations

lewisxhe, 2024-03-12:
You can now see the results.
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/actions/runs/8246524546/job/22552731609

PureTek-Innovations, 2024-03-12:
I just did a fresh download from github.
It still only compiles with the ETH.h library.

PLATFORM: Espressif 32 (2023.10.6+sha.8529a77) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.14
 - tool-esptoolpy @ 1.40700.0+dev2 (4.7.0)
 - tool-mklittlefs @ 3.2.0
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting ETHOTA.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- ETHClass @ 1.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\T-ETH-Lite-ESP32\src\ETHOTA.ino.cpp.o
Compiling .pio\build\T-ETH-Lite-ESP32\lib110\ETHClass\ETHClass.cpp.o
lib/ETHClass/src/ETHClass.cpp: In member function 'bool ETHClass::beginSPI(int, int, int, int, int, int, spi_host_device_t, uint8_t, int, bool)':
lib/ETHClass/src/ETHClass.cpp:314:5: error: 'eth_w5500_config_t' was not declared in this scope
     eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
     ^~~~~~~~~~~~~~~~~~
lib/ETHClass/src/ETHClass.cpp:314:5: note: suggested alternative: 'eth_phy_config_t'
     eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
     ^~~~~~~~~~~~~~~~~~
     eth_phy_config_t
lib/ETHClass/src/ETHClass.cpp:317:5: error: 'w5500_config' was not declared in this scope
     w5500_config.int_gpio_num = irq;
     ^~~~~~~~~~~~
lib/ETHClass/src/ETHClass.cpp:317:5: note: suggested alternative: 'gpio_config'
     w5500_config.int_gpio_num = irq;
     ^~~~~~~~~~~~
     gpio_config
lib/ETHClass/src/ETHClass.cpp:321:15: error: 'esp_eth_mac_new_w5500' was not declared in this scope
     eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
               ^~~~~~~~~~~~~~~~~~~~~
lib/ETHClass/src/ETHClass.cpp:321:15: note: suggested alternative: 'esp_eth_mac_new_esp32'
     eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
               ^~~~~~~~~~~~~~~~~~~~~
               esp_eth_mac_new_esp32
lib/ETHClass/src/ETHClass.cpp:327:15: error: 'esp_eth_phy_new_w5500' was not declared in this scope
     eth_phy = esp_eth_phy_new_w5500(&phy_config);
               ^~~~~~~~~~~~~~~~~~~~~
lib/ETHClass/src/ETHClass.cpp:327:15: note: suggested alternative: 'esp_eth_phy_new_ip101'
     eth_phy = esp_eth_phy_new_w5500(&phy_config);
               ^~~~~~~~~~~~~~~~~~~~~
               esp_eth_phy_new_ip101
*** [.pio\build\T-ETH-Lite-ESP32\lib110\ETHClass\ETHClass.cpp.o] Error 1
================================================================ [FAILED] Took 6.98 seconds

lewisxhe, 2024-03-12:
I'm not sure what's causing this, but you're not the first one using T-ETH-Lite-ESP32, no one else has reported this bug to me, and currently all the checks are passing and I can't to more .

PureTek-Innovations, 2024-03-12:
It may be something to do with Windows 11.
[image]
[image]

lewisxhe, 2024-03-12:
[image]
I also use Win11