Xinyuan-LilyGO/LilyGO-T-ETH-Series #55: LILYGO-T-ETH-LITE S3 Incompatible with Arduino 3 (ESP-IDF 5.1)

mathieucarbou, 2024-03-15:
Add in platformio:

```
platform = espressif32
platform_packages=
  platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-alpha3
  platform_packages=platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
```

And run the tests using this board. IT will fail because it uses the custom `ETHClass` which is not compatible with Arduino 3 (ESP-IDF 5.1).

Would it be possible to provide a working adaptation of it, or us the new implementation in Arduino 3 (ESP-IDF 5.1) ? I see they have a SPIClass that can be implemented for custom boards.

Thanks!

mathieucarbou, 2024-03-15:
I did further research and found some examples using the new framework with the `W5500` adapter which is used in the Lilygo: https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/examples

mathieucarbou, 2024-03-15:
I was able to find a solution, that is implemented in my [MycilaESPConnect](https://oss.carbou.me/MycilaESPConnect/) library (Network manager). I need to support several ETH cards including WT32-ETH1, Lilygo, etc.

Here is what I have in my header:

```cpp
#if defined(ESPCONNECT_ETH_SUPPORT)
  #if defined(ETH_PHY_SPI_SCK) && defined(ETH_PHY_SPI_MISO) && defined(ETH_PHY_SPI_MOSI) &&     defined(ETH_PHY_CS) && defined(ETH_PHY_IRQ) && defined(ETH_PHY_RST)
    #define ESPCONNECT_ETH_SPI_SUPPORT 1
    #if ESP_IDF_VERSION_MAJOR >= 5
      #include <ETH.h>
      #include <SPI.h>
    #else
    #include <ETHClass.h>
    #endif
  #else
    #include <ETH.h>
  #endif
#endif
```

```cpp
#ifdef ESPCONNECT_ETH_SUPPORT
void ESPConnectClass::_startEthernet() {
  _setState(ESPConnectState::NETWORK_CONNECTING);

#if defined(ETH_PHY_POWER) && ETH_PHY_POWER > -1
  pinMode(ETH_PHY_POWER, OUTPUT);
#ifdef ESPCONNECT_ETH_RESET_ON_START
  ESP_LOGD(TAG, "Resetting ETH_PHY_POWER Pin %d", ETH_PHY_POWER);
  digitalWrite(ETH_PHY_POWER, LOW);
  delay(350);
#endif
  ESP_LOGD(TAG, "Activating ETH_PHY_POWER Pin %d", ETH_PHY_POWER);
  digitalWrite(ETH_PHY_POWER, HIGH);
#endif

  ESP_LOGI(TAG, "Starting Ethernet...");
#if defined(ESPCONNECT_ETH_SPI_SUPPORT)
#if ESP_IDF_VERSION_MAJOR >= 5
  // https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/examples
  SPI.begin(ETH_PHY_SPI_SCK, ETH_PHY_SPI_MISO, ETH_PHY_SPI_MOSI);
  if (!ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI)) {
    ESP_LOGE(TAG, "ETH failed to start!");
  }
#else
  if (!ETH.beginSPI(ETH_PHY_SPI_MISO, ETH_PHY_SPI_MOSI, ETH_PHY_SPI_SCK, ETH_PHY_CS, ETH_PHY_RST, ETH_PHY_IRQ)) {
    ESP_LOGE(TAG, "ETH failed to start!");
  }
#endif
#else
  if (!ETH.begin()) {
    ESP_LOGE(TAG, "ETH failed to start!");
  }
#endif

  _lastTime = esp_timer_get_time();

  ESP_LOGD(TAG, "ETH started.");
}
#endif
```

Flags for **wt32-eth01**:

```cpp
  -D ESPCONNECT_ETH_SUPPORT
  -D ETH_PHY_ADDR=1
  -D ETH_PHY_POWER=16
```

Flags for **T-ETH-Lite ESP32 S3**:

```cpp
  -D ESPCONNECT_ETH_SUPPORT
  -D ETH_PHY_ADDR=1
  -D ETH_PHY_CS=9
  -D ETH_PHY_IRQ=13
  -D ETH_PHY_RST=14
  -D ETH_PHY_SPI_MISO=11
  -D ETH_PHY_SPI_MOSI=12
  -D ETH_PHY_SPI_SCK=10
  ; can only be activated with ESP-IDF >= 5
  ; -D ETH_PHY_TYPE=ETH_PHY_W5500
```

mathieucarbou, 2024-03-15:
Oh! And I had to add this hack into `ETHClass` in order to avoid it to be part of the compilation:

```c++
#include "esp_system.h"
#if defined(ETH_PHY_SPI_SCK) && defined(ETH_PHY_SPI_MISO) && defined(ETH_PHY_SPI_MOSI) && defined(ETH_PHY_CS) && defined(ETH_PHY_IRQ) && defined(ETH_PHY_RST) && ESP_IDF_VERSION_MAJOR < 5
```

lewisxhe, 2024-03-16:
At the time of writing, the Arduino examples did not support W5500, I had to write a new class to implement it, I will update it later

mathieucarbou, 2024-03-16:
Yes makes sense. 
I was in the process of preparing my librairies to be compatible.

And like described in this issue it works.

Someone in a normal project won't have to go over all the cases like I had to.

mathieucarbou, 2024-04-21:
I am closing since it is fixed

closed 2024-04-21 by mathieucarbou