Xinyuan-LilyGO/LilyGO-T-ETH-Series #56: T-Internet-POE - Link Up / Down - Loop until freezes

HotNoob, 2024-03-16:
after a few hours of use, the ethernet became unstable, especially after performing OTA updates. 
then shortly after, ethernet completely stopped working. upon startup i get this:

[  9989][V][WiFiGeneric.cpp:431] _arduino_event_cb(): Ethernet Link Up
[  9989][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[  9990][V][WiFiGeneric.cpp:446] _arduino_event_cb(): Ethernet got ip:192.168.1.9
[ 10229][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[ 10230][D][WiFiGeneric.cpp:1141] _eventCallback(): ETH IP: 192.168.1.9, MASK: 255.255.255.0, GW: 192.168.1.1
[ 11989][V][WiFiGeneric.cpp:435] _arduino_event_cb(): Ethernet Link Down
[ 11989][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 21 - ETH_DISCONNECTED

runs for a few times, and then eventually it stops retrying, and the eth port freezes until restarting with lights on or off. 

PoE still works, but no ip connection.

also noticed, mac address randomly changing. mainly the last byte. ie, E4, E7, E2.

tried with ETHClass and Eth.h
most testing was with ETHClass

did my lan chip die / go bad?

lewisxhe, 2024-03-17:
If, as you said, this situation is caused by damage to the LN8720

HotNoob, 2024-03-17:
yeah; i guess so... flashed another one and its been working fine.

closed 2024-03-17 by HotNoob