Xinyuan-LilyGO/LilyGO-T-ETH-Series #58: Use of ETHClass

HungryHungryFish, 2024-04-12:
Every example of ethernet use shows a ETHClass.h and yet for 2 days I get an error fatal error: ETHClass.h: No such file or directory. I'm trying to use the Arduino IDE. I have removed it and reinstalled it. I have added the custom link under preferences and have added the ESP32 library. I have a number of Arduino projects but this is my first ESP32 project. I've restarted, tried chatgpt, google and even bing... any help is appreciated

lewisxhe, 2024-04-13:
Which board to use?

HungryHungryFish, 2024-04-16:
using the supplied board from https://www.aliexpress.com/item/4001122992446.html V2
however I'm just trying to compile in the IDE and getting this error. Not trying to upload to the device yet.

lewisxhe, 2024-04-17:
I updated it a few days ago to support version V3.0. You can try it now. Currently, all Arduino compilations pass.

HungryHungryFish, 2024-04-17:
I'm missing something, sorry. I looked at all the libraries and I don't see anything that I can update. They are all at the latest version. Can you kindly list with libraries I should have? I have verified that Arduino is at the latest version. It appears that I'm missing a library that includes ETHClass.h however everywhere I look I can't see any library that would have it. Thanks for taking the time to respond. Tomorrow I'll try a fresh install of windows and a fresh install of Arduino on another machine to try to rule things out.

lewisxhe, 2024-04-17:
Have you copied all the folders in the lib directory to Arduino according to the home page?
What about the IDE's library directory?

HungryHungryFish, 2024-04-26:
Thanks for responding. After rebuilding the environment on another system the error remained. Turns out I was using the wrong source. Not sure how I grabbed the wrong one but its all good now.

closed 2024-06-10 by github-actions[bot]