Xinyuan-LilyGO/LilyGO-T-ETH-Series #59: Board dont go in download mode

Zkall123, 2024-04-16:
hello my t-eth-lite ESP32 S3 dosnt go in download mode 
when i press the boot butten then press the rst butten release the rst button in the serial monitor of my arduino ide it dosnt show waiting for download.

lewisxhe, 2024-04-17:
What does that show in the Serail monitor? Is the RX TX connected wrong?
The BOOT button should be released after releasing RST. Is the order correct?

Zkall123, 2024-04-17:
[image]
[image]
now there stand waiting for download (second picture) but de code upload dosnt work (First Picture)

lewisxhe, 2024-04-18:
It seems that RX can receive normally and can also enter download mode. Are you using external USB2TTL? I think the problem should be with TX. Which USB2TTL are you using now?

Zkall123, 2024-04-18:
I use this usb2ttl 

https://www.amazon.de/gegenseitige-Konvertierung-Adapter-Multifunktions-USB-Windows/dp/B07XZ3PV8T/ref=mp_s_a_1_2?crid=1JZFOGGV1A7BZ&dib=eyJ2IjoiMSJ9.6PjhhaRHodrdKSo6b4ZbaAsWg8b1tITkJHUarHaZePOBMLoGJbqOpq80LFjlbgJonVvgFIIBbv0a4mEhgKx-i0mhjOnzcsYe03u6JpOuBA3VJ4-ivi9aKzKjBvqJSkVgB9jRIzApvmH-ouFNMJYb6KfQzrH19Ed9oQHhmKjz-hdp-a_VYfBJ4NqWi8bAji5hESlsIoADlNaBSVa7A_K3rA.aQM5hiI_9KsCmyv8dtoqmvYFrEt0ED0TUJ625aoU828&dib_tag=se&keywords=usb+to+serial+adapter+usb2ttl+CP2102%2CCP2104%2CCH340X&qid=1712995104&sprefix=usb+to+serial+adapter+usb2ttl+cp2102+cp2104+ch340x%2Caps%2C95&sr=8-2

lewisxhe, 2024-04-18:
You can short-circuit the RXTX of USB2TTL, and then send the message through the Arduino Serial monitor to see if it can be received.

Zkall123, 2024-04-18:
Okay you mean that i Connect RX to tx in the converter and then send a massage in the serial Monitor?

lewisxhe, 2024-04-18:
Yes, see if it can send and receive normally

Zkall123, 2024-04-18:
Okay thanks i Test IT After Work 

If IT doesnt Work then the converter IS the Problem right?

lewisxhe, 2024-04-18:
yes

closed 2024-04-19 by Zkall123