Xinyuan-LilyGO/LilyGO-T-ETH-Series #60: T-ETH-Lite-S3 all examples fail

troyhacks, 2024-04-17:
This worked fine with the "old" ETHClass.h but none of the examples work now with ETHClass2.h

```
E (9) spi: spi_bus_initialize(756): SPI bus already initialized.
[   165][E][ETHClass2.cpp:452] beginSPI(): spi_bus_initialize failed
ETH start Failed!
```

The SPI bus is initialized twice and fails the second time - not sure if that's a problem - lines 419 and 450. I tried removing one, and the init gets further, but still fails. 

The code seems... somewhat different than ETHClass.h, so I'm curious if there's just a bad commit?

lewisxhe, 2024-04-18:
Which version of ? Arduino IDE? Arduino IDE 3.0-RC1 is tested and verified based on idf 5.1.

lewisxhe, 2024-04-18:
Ok, I only tested 3.0.0-rc1 and forgot to test 2.0.15, now it’s solved

closed 2024-04-18 by lewisxhe