Xinyuan-LilyGO/LilyGO-T-ETH-Series #61: Can't use pin as digital output

EnvioOvi, 2024-04-18:
Hey!

I use an T-ETH-LITE  board and i try to use the digital pin 38 as an output.

The sketch compiles ok, but when the program is running i get the following error:
"E (49528) gpio: gpio_set_level(226): GPIO output gpio_num error"

I get the same error using the pins: 34, 35, 38, 39.
I don't get the error using other pins.

Software used: Arduino IDE 2.2.1, board manager esp32 by Espressif 2.0.11

Can someone help me with a fix on using pin 38 , because the board is done and i can not change the pin.

Thanks!

kimble4, 2024-04-18:
Aren't those pins input-only on an ESP32?

https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/gpio.html

EnvioOvi, 2024-04-18:
:( i guess so ...
Thank you!

closed 2024-06-02 by github-actions[bot]