Xinyuan-LilyGO/LilyGO-T-ETH-Series #64: T-POE Pro pinout

MosquitoByte, 2024-05-01:
Hello,
I am considering the T-POE Pro for a project.
However the pinout marking for SPI is unclear.
Can please you provide a pinout diagram for this board?
Thank you

lewisxhe, 2024-05-02:
[pinmap](https://www.lilygo.cc/products/t-poe-pro)
SPI pin can be freely mapped to unused GPIO, please refer to https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/examples/SPIExample/SPIExample.ino

MosquitoByte, 2024-05-02:
Thank you for your reply.
Please have a look at the attached annotated picture.
What is the pinout of the pins circled in red?
Where are the corresponding pins of those described in the green circle?
Thanks
[image]

lewisxhe, 2024-05-09:
The green sequence corresponds to the pin row below

lewisxhe, 2024-05-09:
[image]
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/schematic/T-POE-PRO.pdf

closed 2024-06-24 by github-actions[bot]

solexious, 2024-08-07:
> ![image](https://private-user-images.githubusercontent.com/22990954/329192715-4ec788e2-bd05-40f8-8f59-0f8d1bb0d952.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwMjk5MTcsIm5iZiI6MTcyMzAyOTYxNywicGF0aCI6Ii8yMjk5MDk1NC8zMjkxOTI3MTUtNGVjNzg4ZTItYmQwNS00MGY4LThmNTktMGY4ZDFiYjBkOTUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDExMjAxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ2ZmM1NTE5Y2FmOTllMWNlZTdhNWUwYzg2MThhYjNiMzBmMTcxNzg1ODMwMjA1ZTBhZGRlMDk2ZGY3YTg3N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qo9DLRINzLEgs5a0gzNabv1W6w4pCIv6eit3wciQdxc) https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/schematic/T-POE-PRO.pdf

Please consider adding this to the promotional photos that show pin outs, I had to search for this information for quite a while to find it.