Xinyuan-LilyGO/LilyGO-T-ETH-Series #67: T-ETH-Pro ModBus issues

MosquitoByte, 2024-05-22:
Hello,
Using esphome on the T-ETH-Pro to read information over ModBus.
When defining RS485 pins as published (TX:IO32, RX:IO33) no information is received.
When switching TX/RX pins, information is received but with echoes and CRC errors.
Testing the same setup with other ESP32 board+Max3485 chip gives perfect results - so the problem is not wiring or termination.
Any idea what should I try?

lewisxhe, 2024-05-24:
1. What is the baud rate of the communication speed?
2. How many meters is the communication distance?
3. Have you done any sending and receiving tests between the two boards? Will data errors occur?

MosquitoByte, 2024-05-24:
1. It's 9600 baud.
2. Very short, 1 device connected with a 2 meter STP cable.
3. Didn't try connecting the two boards, however when I debug the data on the UART I see that every command is sent but an echo is also immediately received, as if the bus is not turning RX off while transmitting as it should be.

Thanks, and I hope we can sort it out since we are testing this setup for a 2000 pieces project.

lewisxhe, 2024-05-24:
The occurrence of the loopback phenomenon has a certain relationship with the automatic control of sending and receiving. The chip's receiver is enabled at other times except when it is in a disabled state when sending low levels.

Do you have any resistors? You can try adding a 10K pull-up resistor to the first pin of the MAX13487E chip to the eighth pin of the MAX13487E chip. See if the situation improves.

MosquitoByte, 2024-05-24:
Please provide a picture or schematic of the required modification. First pin and eighth pin is a bit ambiguous.
 Thanks.

lewisxhe, 2024-05-24:
[image]

MosquitoByte, 2024-05-24:
Thanks, I will try and let you know

MosquitoByte, 2024-06-04:
After connecting a 10k resistor as instructed everything works perfectly.
Is this a design flaw and you will put this resistor on production?

lewisxhe, 2024-06-05:
It has been submitted to the hardware engineer and will be added later.

Llgok, 2024-06-05:
> 按照说明连接 10k 电阻器后，一切正常。这是一个设计缺陷，你会把这个电阻器投入生产吗？

@MosquitoByte 
Hello,

Here is the response to your question regarding the MAX13487E/MAX13488E series chips. The "RE" pin is used to control the device's AutoDirection and the use of callback functions. By default, on the PCB board, a 10K pull-up resistor is used to enable AutoDirection and disable callback functions. This is not a design flaw; the module's automatic data transmission and reception do not require manual control over data output and input.

closed 2024-07-19 by github-actions[bot]

tbnobody, 2024-07-19:
If someone else faces this issue, just solder a 10k 0805 resistor between this two pins:
[image]

Works perfectly after this modification!