Xinyuan-LilyGO/LilyGO-T-ETH-Series #68: T-ETH-Lite: Second SPI for other device than TF Card?

JFLSeoul, 2024-05-22:
Hi, I want to use the W5500 and on the other SPI interface I want to connect a RFID reader and I do not need to use the TF Card. Is this still possible with the Hardware Layout ? Thank you very much !

lewisxhe, 2024-05-24:
Of course. ESP can map any GPIO to SPI, you just need to specify the Pin during initialization.

JFLSeoul, 2024-05-24:
Thank you for your answer. I wanted to use HSPI and VSPI for hardware SPI, which most esp32 boards have. So I thought that the W5500 might be using dedicated HSPI pins and the TF card might be using dedicated VSPI pins. What you suggest sounds a bit like software SPI?

lewisxhe, 2024-05-25:
Didn't you say you don't need to use an SD card?  You don't have to worry about the pins of the SD card. For other SPI devices, you only need to specify GPIO initialization like the SD card. Of course, you can also reuse SPI. This is used in many LilyGo devices using the SPI bus.

JFLSeoul, 2024-05-25:
Yes, you are right, I do not want to use the SD card. On HSPI I want to use the w5500 and on VSPI I want to connect a RFID reader ( instead of the SD card). So to me it sounds possible and I will try:) thank you very much !

closed 2024-07-09 by github-actions[bot]