Xinyuan-LilyGO/LilyGO-T-ETH-Series #69: T-ETH-Lite-POE-Shield : Isolation?

fplanque, 2024-05-23:
Does the [T-ETH-Lite-POE-Shield](https://www.lilygo.cc/products/t-eth-lite?variant=43120887038133) provide **electrical isolation** between Ethernet and the ESP32, I/O, USB, etc ?

I can't find a datasheet that gives this information.

Electrical isolation is important. Here are quotes from similar product datasheets:

- wESP32 datasheet: "The board is designed to provide electrical isolation between the ESP32 application portion and the Ethernet cable to ensure safety and performance as demanded by the IEEE 802.3at specification."
- ESP32-POE-ISO: "comes with 3000VDC galvanic insulation between the Ethernet POE part and ESP32 power supply, so there is no problem to connect to USB and program or debug while the board is also powered from a PoE source."

And on the **contrary** for a NON isolated board: "YOU SHOULD NOT CONNECT ESP32-POE to computer’s USB port while it is powered by Ethernet POE!!! If you connect USB while ESP32-POE is powered by Ethernet you will damage the board or your computer or both. This will also void the warranty!"

So, what's the isolation level of the T_ETH-Lite-POE-Shield ?

Thank you.

lewisxhe, 2024-05-24:
[image]

fplanque, 2024-05-24:
@lewisxhe Thank you for this screenshot. 
Where does it come from? 
Can you please link to the original source?

lewisxhe, 2024-05-25:
http://sdapo.com/show/56.html

closed 2024-07-09 by github-actions[bot]