Xinyuan-LilyGO/LilyGO-T-ETH-Series #72: No serial data received

programmeddeath1, 2024-06-17:
I have ordered the T-ETH-lite esp32-s3 module. 

The board is powering on and the tx for the upload gets activated, but no data is received from the board.

I have tried with a usb to ttl with [FT232RL](https://www.electronicscomp.com/waveshare-industrial-usb-to-ttl-converter-original-ft232rl?gad_source=1&gclid=CjwKCAjwmrqzBhAoEiwAXVpgopN5u46hR2ihEh264OZTCTT9yuNtUHnDgsmL48g-6YNcR4Ag7nC0VxoCNRUQAvD_BwE) as well as with [PL2303](https://robu.in/product/pl2303-ta-download-cable-usb-ttl-rs232-module-usb-serial/?gad_source=1&gclid=CjwKCAjwmrqzBhAoEiwAXVpgojE0Hci680cSpoTpx4LIlNwWa5VSJ9t9IGKAy1luo6SeN3GPgEZg2RoCX0QQAvD_BwE)

Both are giving me the same issue.

The power led is always on and the ACT led(green) is on and blinks intermittently. When I hold BOOT and then press RST, the link led(Yellow) turns on for sometime and then turns off. 

But the data does'nt seem to be getting received from the esp32-s3, possibly it isnt even transmitting to the esp32, but the waeveshare convertor shows the tx activated when uploading to the board.

What could be the issue, is the board damaged or do i need to try with a CP2102 or 2104 connector?

lewisxhe, 2024-06-19:
Using Arduino IDE? Set `CDC On Boot` to `Disable`

programmeddeath1, 2024-06-19:
Yeah I am running through arduino ide, all those settings are set as shown in the README. I was able to get it to work with the PL2303, usb to ttl.

It just does'nt detect with the FT232RL waveshare usb to ttl convertor, tried many reset but the waiting to download or even device details do not show up. I think I will have to try with the recommended CP2102 connector.

closed 2024-06-19 by programmeddeath1

Yibing-Chen, 2025-01-14:
I found some cheap CHARGE ONLY usb cable may cause this problem...