Xinyuan-LilyGO/LilyGO-T-ETH-Series #74: T-ETH-Lite-S3 W5500 - Can it go faster?

troyhacks, 2024-07-23:
I'm curious if there's any areas for speed improvement with the W5500 code?

My same code (obviously a different driver) running on a board with a LAN8720 has about a 3x increase in transfer speed (~10mbit on the W5500 vs ~30mbit on the LAN8720)

The T-ETH-Lite-S3 even transmits faster on its internal WiFi than using the onboard W5500 - although the W5500 is much more stable.

I am using `-D ETH_PHY_SPI_FREQ_MHZ=60` which provides some benefit over the default 20MHz setting. (80Mhz does not work in my testing)

The W5500 code works great and is super stable - I was just surprised with the speed testing being so slow.

troyhacks, 2024-07-23:
Here I start with the onboard WiFi and plug in the Ethernet:

```
Setup took    5 micros. UDP for 8064 pixels took 17614 micros. 25056 data in 48 total packets. 11.95 mbit/sec at 38 FPS.
Setup took    6 micros. UDP for 8064 pixels took 16085 micros. 25056 data in 48 total packets. 13.08 mbit/sec at 38 FPS.
Setup took    6 micros. UDP for 8064 pixels took 16754 micros. 25056 data in 48 total packets. 12.56 mbit/sec at 38 FPS.
Network Event: ARDUINO_EVENT_ETH_CONNECTED = ETH connected. Setting up ETH
Ethernet speed is 100 mbit and link is full duplex! (MAC: 36:85:18:8F:40:25)
Network Event: ARDUINO_EVENT_ETH_GOT_IP = ETH is up. Disabling WIFi
Setup took    5 micros. UDP for 8064 pixels took 19867 micros. 25056 data in 48 total packets. 10.59 mbit/sec at 34 FPS.
Setup took    5 micros. UDP for 8064 pixels took 19866 micros. 25056 data in 48 total packets. 10.59 mbit/sec at 34 FPS.
Setup took    7 micros. UDP for 8064 pixels took 19713 micros. 25056 data in 48 total packets. 10.67 mbit/sec at 34 FPS.
```

With the LAN8720 it's running at around 30 mbit/sec and my FPS is correspondingly much higher.

troyhacks, 2024-07-30:
Recompiled arduino-esp32 with some optimizations and non-default options enabled for the ESP32-S3 and the W5500 bandwidth does improve. 

```
UDP for 8064 pixels took 17118 micros. 25056 data in 48 total packets. 12.29 mbit/sec at 40 FPS.
UDP for 8064 pixels took 17087 micros. 25056 data in 48 total packets. 12.31 mbit/sec at 40 FPS.
UDP for 8064 pixels took 17102 micros. 25056 data in 48 total packets. 12.30 mbit/sec at 40 FPS.
```

Microseconds for 48 packets is down from ~19800 to ~17100 👍🏻

closed 2024-09-13 by github-actions[bot]

programmeddeath1, 2024-10-14:
Hii @troyhacks  can you elaborate a little on what options you changed to be able to get this improvement as well as the arduino-esp32 optimizations? would be quite helpful. I am trying to send data over ethernet to my esp32 at as low a latency as possible. I am trying to send <100 byte data at a frequency of 10ms. Over websocket on ETH8710A on the ESP32-EVB olimex board, I was getting round-trip times of around 3ms. But on the T-ETH-LITE S3, its giving consistent times of around 7ms. I feel like it can go faster since the hardware clearly has more room to work on, but we could be limited by the fact that the arduino-esp is not as optimized for the esp32 as the esp-idf components

troyhacks, 2024-10-14:
> Hii @troyhacks  can you elaborate a little on what options you changed to be able to get this improvement as well as the arduino-esp32 optimizations? would be quite helpful.

I'm a really dumb smart person so you'll have to excuse the lack of technical details:

I recompiled it with -O2 and also added instruction buffers and told it to put the SPI functions into IRAM. These are all "menuconfig" options you can set.

Basically I effed around with the build until it was faster. 😁

The limitation here is the SPI. The ESP32 and the new ESP32-P4 have RMII Ethernet and it's much faster than SPI. (The ESP32-P4 is about 2x faster than the original ESP32 over Ethernet in my experiments.)

programmeddeath1, 2024-10-14:
understood, will tinker around with the builds thats something that always brings about some results. 

The gains on the P4 are huge. It makes sense that the SPI is the bottleneck. For me it has to do with the easy availability of the boards for my controllers. this board is available easily in case we need to meet customer SLA's so kinda stuck with it.