Xinyuan-LilyGO/LilyGO-T-ETH-Series #79: Example: CameraShield.ino it does not compile

shojkeee, 2024-08-14:
Hi. During the compilation of the sketch, CameraShield.ino Throws an error that there is not enough memory to load. Please tell me what can be removed from the sketch so that there is enough space?
Or maybe my versions of lib are not like that? and do they take up more space as a result?

I have a T-ETH-Lite ESP32S3 board and a T-ETH-Lite Shield Camera

[image]

shojkeee, 2024-08-16:
> https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/16cf3f291f34a10a024d3ba4b00854e796781c33/examples/CameraShield/CameraShield.ino#L27

I choose 3MB and still don't have enough space to upload to the board

[image]

lewisxhe, 2024-08-16:
Strange, it was working fine before, now I put a custom partition and this problem doesn't exist anymore.

lewisxhe, 2024-08-16:
[image]

shojkeee, 2024-08-16:
> ![image](https://private-user-images.githubusercontent.com/22990954/358531378-1a8d7966-a5ba-4af3-906b-fca024a5e1cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3OTk0NzcsIm5iZiI6MTcyMzc5OTE3NywicGF0aCI6Ii8yMjk5MDk1NC8zNTg1MzEzNzgtMWE4ZDc5NjYtYTViYS00YWYzLTkwNmItZmNhMDI0YTVlMWNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE2VDA5MDYxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkZDYyMTcxZmJlNmFjZjM1Mzk5YTY3MzZlYmRmYTQ1NjA4MDRlYzZjYWU1YTEzNDM0MzIzOGU1ZmRlYTNjYjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GJ7BL6ZNv1rNJQvEmftSEmCNAUeaZVYaXYqWwKHiweM)

And how did you do it? I don't have a "custom"

lewisxhe, 2024-08-16:
Update esp32 to the latest version in the board manager

shojkeee, 2024-08-16:
> Update esp32 to the latest version in the board manager

I'm probably kind of unlucky. :(. Did as you said, updated esp32. I created the partition.csv file, but it still gives an error when compiling.

Traceback (most recent call last):
  File "gen_esp32part.py", line 612, in <module>
  File "gen_esp32part.py", line 565, in main
  File "gen_esp32part.py", line 163, in from_file
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbe in position 12: invalid start byte
[19256] Failed to execute script 'gen_esp32part' due to unhandled exception!

[image]

lewisxhe, 2024-08-17:
I asked my colleague to test it, and he was able to compile it successfully. You can try to delete esp32 and reinstall it to test it.

shojkeee, 2024-08-17:
> I asked my colleague to test it, and he was able to compile it successfully. You can try to delete esp32 and reinstall it to test it.

The fact is that if I selected Other parameters in the "Prtition scheme" item, where 4 MB was allocated. And I compiled everything perfectly with these parameters, too. But when loading the sketch into the board, it went into a cyclic reboot.

I'm still at work. I work for days. As soon as I get home, I'll check your option.

shojkeee, 2024-08-17:
> I asked my colleague to test it, and he was able to compile it successfully. You can try to delete esp32 and reinstall it to test it.

No, it doesn't work. The fee is looping

mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdad1
SPIWP:0xee
mode:DIO, clock div:1

lewisxhe, 2024-08-17:
I will test again on Monday, it's a holiday now

shojkeee, 2024-08-17:
> I will test again on Monday, it's a holiday now

Thank you. I'll be waiting. Have a nice weekend

lewisxhe, 2024-08-19:
Hi, I tested it and it works fine. Can you check if the settings are correct?
[image]

shojkeee, 2024-08-19:
> Hi, I tested it and it works fine. Can you check if the settings are correct? [...]

Yes. the board settings are the same, I took them from the description. Have you only checked the compilation? Or have you downloaded the program and see the image from the camera?

lewisxhe, 2024-08-19:
Of course, normal
[image]

closed 2024-10-03 by github-actions[bot]