Xinyuan-LilyGO/LilyGO-T-ETH-Series #81: How to connect to LTE modem using USB?

ruslanslobozhenuk, 2024-08-20:
I received an A7670E modem. How can I connect to it using USB?

lewisxhe, 2024-08-22:
Turn the USB switch on the back of the LTE Shiled to ON, and you can connect the USB of the MODEM and ESP together. Please note that if you do this, you cannot insert the TYPEC to write the firmware. You must use the firmware before inserting it, or you use URAT 2 TTL to connect to the 40PIN
pin to access the ESP

lewisxhe, 2024-08-22:
Here is a picture
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/tree/master/esp-idf-examples/usb_cdc_4g_module

ruslanslobozhenuk, 2024-08-22:
Thank you!
Tell me where I can buy a USB to UART adapter with cable, like the one in the example picture?
Thanks in advance!

closed 2024-10-06 by github-actions[bot]