Xinyuan-LilyGO/LilyGO-T-ETH-Series #83: I2C connection ETH-LITE S3

netsystemsrl, 2024-08-30:
I have Eth-liste s3,
I'havent find in map and in documentation wich pin I have to use to I2C

but I find in github sample: 
#elif defined(LILYGO_T_ETH_LITE_ESP32S3) 
// ESP32S3 can freely map unused Pins 
**#define I2C_SDA 1 
#define I2C_SCL 2** 

I have try with this code but, nothing is discovered... I have try also with some i2c component... but still no response... 
can y help me ?
thanks

lewisxhe, 2024-08-30:
Paste all your test code

closed 2024-10-15 by github-actions[bot]

netsystemsrl, 2024-11-29:
I have define:

#define SDA_PIN 1
#define SCL_PIN 2
Wire.begin(SDA_PIN, SCL_PIN);

scanI2CDevices(&Wire, &Serial);

uint32_t scanI2CDevices(TwoWire *_port, Stream *stream){
stream->println("Devices Scan start.");
uint8_t err, addr;
int nDevices = 0;
for (addr = 1; addr < 127; addr++) {
_port->beginTransmission(addr);
err = _port->endTransmission();
if (err == 0) {
stream->print("I2C device found at address 0x");
if (addr < 16)
stream->print("0");
stream->print(addr, HEX);
stream->println(" !");
nDevices++;
} else if (err == 4) {
stream->print("Unknow error at address 0x");
if (addr < 16)
stream->print("0");
stream->println(addr, HEX);
}
}
if (nDevices == 0)
stream->println("No I2C devices found\n");
else
stream->println("Done\n");
return nDevices;
}

I receive only No I2C devices found!

lewisxhe, 2024-11-30:
Lite does not have any i2c components, have you connected the module?