Xinyuan-LilyGO/LilyGO-T-ETH-Series #85: Use the T-ETH (ETH-Lite)with IDF 5.3  ethernet examples

MrutyunjayPatel, 2024-09-09:
hi,
I need the hardware to compile and run  with idf version 5.xx i want o use IDF5.14 and IDF5.3x .

After selecting example project  "D:\esp-idf-v5.3.X\examples\ethernet\basic"  

idf.py set-target esp32 
idf.py menuconfig 

Selected following options 

(Top) → Example Ethernet Configuration        Espressif IoT Development Framework Configuration                                   
 [*] Internal EMAC
        Ethernet PHY Device (RTL8201/SR8201)  --->
(23)    SMI MDC GPIO number
(18)    SMI MDIO GPIO number
(-1)    PHY Reset GPIO number
(0)     PHY Address
[ ] SPI Ethernet          

After building and flashing and in monitor i get following eth init "reset timeout errors"

```
E (2608) esp.emac: emac_esp32_init(426): reset timeout
E (2608) esp_eth: esp_eth_driver_install(228): init mac failed
E (2608) example_eth_init: eth_init_internal(100): Ethernet driver install failed

```
Tried to both phy address  0 and 1 and also -1 (autodetect) always gets   

Did I miss any other settings to be set for this board to work with IDF projects . 

-----Monitor errors ------
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (31) boot: compile time Sep  9 2024 12:12:02
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=12de8h ( 77288) map
I (131) esp_image: segment 1: paddr=00022e10 vaddr=3ffb0000 size=02474h (  9332) load
I (135) esp_image: segment 2: paddr=0002528c vaddr=40080000 size=0ad8ch ( 44428) load
I (155) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=310bch (200892) map
I (224) esp_image: segment 4: paddr=000610e4 vaddr=4008ad8c size=02a98h ( 10904) load
I (235) boot: Loaded app from partition at offset 0x10000
I (235) boot: Disabling RNG early entropy source...
I (247) cpu_start: Multicore app
I (256) cpu_start: Pro cpu start user code
I (256) cpu_start: cpu freq: 160000000 Hz
I (256) app_init: Application information:
I (259) app_init: Project name:     ethernet_basic
I (264) app_init: App version:      1
I (268) app_init: Compile time:     Sep  9 2024 12:55:34
I (274) app_init: ELF file SHA256:  bf88a08c2...
I (280) app_init: ESP-IDF:          -128-NOTFOUND
I (285) efuse_init: Min chip rev:     v0.0
I (290) efuse_init: Max chip rev:     v3.99
I (295) efuse_init: Chip rev:         v3.1
I (300) heap_init: Initializing. RAM available for dynamic allocation:
I (307) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (313) heap_init: At 3FFB3748 len 0002C8B8 (178 KiB): DRAM
I (319) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (325) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (332) heap_init: At 4008D824 len 000127DC (73 KiB): IRAM
I (340) spi_flash: detected chip: generic
I (343) spi_flash: flash io: dio
I (348) main_task: Started on CPU0
I (358) main_task: Calling app_main()
W (358) eth_example: -----IDF5.3x-------Ethernet CHIP RTL8201 ADDR=00 RST_GPIO= -1 -------------
E (2608) esp.emac: emac_esp32_init(426): reset timeout
E (2608) esp_eth: esp_eth_driver_install(228): init mac failed
E (2608) example_eth_init: eth_init_internal(100): Ethernet driver install failed
I (2618) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2618) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2628) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2638) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2648) gpio: GPIO[22]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2658) gpio: GPIO[23]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2668) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2678) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2688) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (2698) example_eth_init: example_eth_init(260): internal Ethernet init failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d7e56
0x400d7e56: app_main at D:/ESP32-WorkspaceIDF53X/basic/main/ethernet_example_main.c:79 (discriminator 1)

file: "./main/ethernet_example_main.c" line 79
func: app_main
expression: example_eth_init(&eth_handles, &eth_port_cnt)

abort() was called at PC 0x40086033 on core 0
0x40086033: _esp_error_check_failed at D:/esp-idf-v5.3.X/components/esp_system/esp_err.c:49

Backtrace: 0x400819aa:0x3ffb58d0 0x4008603d:0x3ffb58f0 0x4008c069:0x3ffb5910 0x40086033:0x3ffb5980 0x400d7e56:0x3ffb59b0 0x40100823:0x3ffb5a40 0x400869cd:0x3ffb5a70
0x400819aa: panic_abort at D:/esp-idf-v5.3.X/components/esp_system/panic.c:463
0x4008603d: esp_system_abort at D:/esp-idf-v5.3.X/components/esp_system/port/esp_system_chip.c:92
0x4008c069: abort at D:/esp-idf-v5.3.X/components/newlib/abort.c:38
0x40086033: _esp_error_check_failed at D:/esp-idf-v5.3.X/components/esp_system/esp_err.c:49
0x400d7e56: app_main at D:/ESP32-WorkspaceIDF53X/basic/main/ethernet_example_main.c:79 (discriminator 1)
0x40100823: main_task at D:/esp-idf-v5.3.X/components/freertos/app_startup.c:208
0x400869cd: vPortTaskWrapper at D:/esp-idf-v5.3.X/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: bf88a08c2

Rebooting...
```
-----

MrutyunjayPatel, 2024-09-09:
also flashed the unit test firmware 
board is ok

lewisxhe, 2024-09-10:
[image]

programmeddeath1, 2024-10-14:
Hi can you guide us to how we can get ethernet or websockets working on the T-ETH-LITE ESP32S3, using esp-idf? If you guys will take some time to enable its examples can you provide a guide to how users can get it done themselves if they want to start on it? Any help would be appreciated.

closed 2024-11-29 by github-actions[bot]