Xinyuan-LilyGO/LilyGO-T-ETH-Series #90: proper pinout documentation!

maskedfighter, 2024-11-07:
Hi!

Can u please do a proper pinout documentation so we can start programming esphome without having to search for each and every single pin we might need to use? Honestly it´s a pain in the ass... 
For example, where do i find which power pin for ethernet i have to use on my t poe pro board?

Thx in advance!

lewisxhe, 2024-11-07:
Thanks for your advice, it's a good suggestion, it will take a little time to do.

The pinhole pointed to in the picture should be the 5V power pin
[image]

maskedfighter, 2024-11-07:
This is the part of code i used to configure the ethernet port in esphome:
ethernet:
  type: LAN8720
  mdc_pin: 23
  mdio_pin: 18
  clk_mode: GPIO0_OUT
  phy_addr: 0
  power_pin: 5  
  # Optional fallback IP configuration
  
  Now when i try using external power to the board (not poe), the port doesn´t come to live, lights on port stay black and nothing happens...
  Any ideas?

lewisxhe, 2024-11-07:
Is the USB power supply normal?

maskedfighter, 2024-11-07:
I don´t supply power to the usb-c port but to the terminal on the right side of the ethernet port (7-24V terminal)

lewisxhe, 2024-11-07:
I mean, does it work fine using USB-C for power delivery?

maskedfighter, 2024-11-07:
Have to try that, one second :)

lewisxhe, 2024-11-07:
With diodes for isolation, they should be able to supply power at the same time. The end with the highest voltage should be the final power supply.
[image]

maskedfighter, 2024-11-07:
aah i see...  so it will probably powered by poe since its voltage is likely higher than the 12V i supply on the terminal?

closed 2024-12-23 by github-actions[bot]