Xinyuan-LilyGO/LilyGO-T-ETH-Series #91: I2C pins on T-Internet-POE

huub66, 2024-11-12:
Which pins on the T-Internet-POE (Lilygo PoE v1.0) board correspond for an I2C bus (sda/scl) for use with ESPHome?

lewisxhe, 2024-11-13:
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/bc0b085d3777403d4a6e51d795cac48c6c4314c6/examples/I2C_BME280Sensor/I2C_BME280Sensor.ino#L27

huub66, 2024-11-13:
> https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/bc0b085d3777403d4a6e51d795cac48c6c4314c6/examples/I2C_BME280Sensor/I2C_BME280Sensor.ino#L27

I indeed noticed this exaple, but where can I find pin 14 & 15 on the board?

netsystemsrl, 2024-11-29:
I have define:

#define SDA_PIN 1
#define SCL_PIN 2
Wire.begin(SDA_PIN, SCL_PIN);

scanI2CDevices(&Wire, &Serial);

uint32_t scanI2CDevices(TwoWire *_port, Stream *stream){
    stream->println("Devices Scan start.");
    uint8_t err, addr;
    int nDevices = 0;
    for (addr = 1; addr < 127; addr++) {
        _port->beginTransmission(addr);
        err = _port->endTransmission();
        if (err == 0) {
            stream->print("I2C device found at address 0x");
            if (addr < 16)
                stream->print("0");
            stream->print(addr, HEX);
            stream->println(" !");
            nDevices++;
        } else if (err == 4) {
            stream->print("Unknow error at address 0x");
            if (addr < 16)
                stream->print("0");
            stream->println(addr, HEX);
        }
    }
    if (nDevices == 0)
        stream->println("No I2C devices found\n");
    else
        stream->println("Done\n");
    return nDevices;
}

I receive only No I2C devices found!

closed 2025-01-13 by github-actions[bot]