Xinyuan-LilyGO/LilyGO-T-ETH-Series #92: T-ETH-Elite Gateway

koeckm, 2024-11-14:
Hi there,

I recently got my T-ETH-Elite with the LoRaWAN-Shield, the LoRaWAN-Module(SX1302), the LTE-Shield, and the LTE-Modul (A7670E). Now I want to build a Gateway which recieves the LoRa-Data and sends it via LTE to the TTN Cloud. Can someone help me how to set it up, because i watched the LilyGo YouTube video of how to use the LoRaWAN-Shiel and even there i was not able to get it running.

Kind regards Markus

lewisxhe, 2024-11-15:
Please let me know which step you encountered the problem

koeckm, 2024-11-15:
Meanwhile i was able to connect the T-ETH-ELITE to the TTN, but i think in the video description is the wrong code because i needed the firmware of the T-ETH-Elite Gateway. My problem now is that cant see any GPS data on the TTN and that i am using WiFi to transmitt data to TTN instead of the LTE.

lewisxhe, 2024-11-15:
There is no relevant code for LTE at present. Currently only ETH or WIFI are supported.

koeckm, 2024-11-15:
The Switches were already set to ON, but in the Serial Debug Assistant it always says that there are no valid GPS coordinates yet even though i have GPS coordinates with my mobile phone.

In what period can i expect the code to support the LTE-Sield in combination with the LoRa-Shield to be released?

lewisxhe, 2024-11-16:
Is the antenna installed? Is the antenna placed outdoors or is the device outdoors?

No updates are planned at this time

koeckm, 2024-11-17:
I will try that.

And there is no other way to get the LTE Shield running?

lewisxhe, 2024-11-18:
LTE Shield example https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/tree/master/esp-idf-examples/usb_cdc_4g_module

koeckm, 2024-11-25:
How can I flash this code on the ESP32? I have tried it with Visual Code and the ESP-IDF extension but I was not able to connect. I could not see the SSID either. Do I have to flash it with the QWIIC or is it possible to flash it with USB-C?

lewisxhe, 2024-11-26:
You must use the UART on the Qwiic to view the serial. You cannot connect USB-C because the USB-C is already connected to the modem.

closed 2025-01-09 by github-actions[bot]