Xinyuan-LilyGO/LilyGO-T-ETH-Series #93: T-POE Pro stuck on "waiting for download" at first power on or after programming

ddv2005, 2024-12-09:
Hello,

I have huge problem with boot stability of T-POE Pro. After each firmware loading I have to manually reset board by reset button because it went to "waiting for download" mode after reset. Also it stuck at "waiting for download" when I connect Ethernet cable with POE. Board absolutely unusable because can't properly boot after power loss.

lewisxhe, 2024-12-09:
Take a video of the video still in download mode after reset, send it to the dealer, they will confirm and replace it with you

NetForces, 2024-12-10:
I also have this issue with the T-ETH-Lite ESP32. Not always, but I need to press reset multiple time to get it to boot and not go in "waiting for download" mode.

NetForces, 2025-01-07:
Here is an example where after uploading code I had to press the reset button 6 times before the ESP-32 got out of the "waiting for download" mode:

```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Starting...
Firmware version: v0.0.2
```

Any resolution for this?

lewisxhe, 2025-01-09:
The PHY clock is output to GPIO0, so there is a certain probability that the clock output will affect the download mode during startup. This problem cannot be solved for the time being. The same situation applies to boards that use IO0 as the clock input.

NetForces, 2025-01-09:
Ok. I will integrate an external watchdog to fix this.

closed 2025-02-23 by github-actions[bot]

NetForces, 2025-02-23:
For those interested, to "fix" this, what I did was add an external watchdog circuit with a delay of 5 minutes. So if the watchdog is not triggered for longer than 5 minutes, the board will reset and hopefully boot on the code and not in download mode. If it does again go in download mode then again in the next 5 minutes it will get reset again.

I used am ATTiny85 for the watchdog based on this: https://github.com/sktaylortrash/atTiny85-Watchdog