Xinyuan-LilyGO/LilyGO-T-ETH-Series #94: esp-idf for T-POE Pro

yltsa, 2024-12-12:
Hi,

Please add esp-idf files for T-POE Pro. Professional hw needs also professional embedded tools.

BR, Ilkka

lewisxhe, 2024-12-13:
There is no plan to add it for now. I will add examples of each idf version later.

closed 2025-01-28 by github-actions[bot]

yltsa, 2025-02-03:
Hi, I created an esp-idf project. Now the problem is that I need to have some Esp32-S3 based board for the platformio.ini file.

By default people uses

- platform = espressif32
- board = esp32dev

That board does not have native USB-OTG support. I need S3 based board. esp32-s3-devkitc-1 is S3 based it but I cannot configure the PHY/Etherfnet for it. 

Is there any board which contains USB-OTG and PHY/Ethernet?