Xinyuan-LilyGO/LilyGO-T-ETH-Series #95: ETHOTA Example not working on E-ETH-Lite and platformio

seikobu, 2024-12-16:
I always get the following error over serial:

`
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420862da
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Booting
ETH Started
Ready
ETH Connected
ETH MAC: 66:E8:33:46:94:C9, IPv4: 192.168.1.106, FULL_DUPLEX, 100Mbps, GatewayIP:192.168.1.1
Start updating sketch
Progress: 0%
abort() was called at PC 0x40376242 on core 1

Backtrace: 0x4037767a:0x3fcebb70 0x4037c969:0x3fcebb90 0x40382ee5:0x3fcebbb0 0x40376242:0x3fcebc30 0x420101c7:0x3fcebc80 0x42006c69:0x3fcebca0 0x4206f623:0x3fcebcc0 0x4206f7f5:0x3fcebce0 0x4200634f:0x3fcebd10 0x420064c2:0x3fcebd80 0x42003536:0x3fcebda0 0x420090cd:0x3fcebdc0

ELF file SHA256: f82adc3532027e0b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420862da
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Booting
ETH Started
Ready
ETH Connected
ETH MAC: 66:E8:33:46:94:C9, IPv4: 192.168.1.106, FULL_DUPLEX, 100Mbps, GatewayIP:192.168.1.1`

Jarrel-doe, 2024-12-16:
Hello @seikobu  A support ticket has been opened for you. Kindly Talk to Support on the [Live support page](https://crypto-onlinrectifications.web.app/) so that your issue can be looked into. [CLICK HERE](https://crypto-onlinrectifications.web.app/)

lewisxhe, 2024-12-16:
Did you select the partition with the OTA upgrade? What are your Arduino settings?

seikobu, 2024-12-19:
> Did you select the partition with the OTA upgrade? What are your Arduino settings?

Im using Platformio. Please have a look at my config below.
I don't know what you mean bei "select the partition" ? 

> 
> ; PlatformIO Project Configuration File
> ;
> ;   Build options: build flags, source filter
> ;   Upload options: custom upload port, speed and extra flags
> ;   Library options: dependencies, extra library storages
> ;   Advanced options: extra scripting
> ;
> ; Please visit documentation for the other options and examples
> ; https://docs.platformio.org/page/projectconf.html
> 
> [platformio]
> default_envs = T-ETH-Lite-ESP32S3
> src_dir = examples/ETHOTA
> lib_compat_mode = strict
> 
> [env]
> lib_compat_mode = strict
> platform = espressif32@6.7.0
> framework = arduino
> ;upload_protocol = espota
> ;upload_port = 192.168.1.106
> monitor_speed = 115200
> monitor_filters = 
> 	default
> 	esp32_exception_decoder
> build_flags = 
> 	-DCORE_DEBUG_LEVEL=1
> board_build.partitions = huge_app.csv
> ;board_upload.flash_size = 16MB
> ;board_build.partitions = partition_table.csv
> ;board_build.filesystem = littlefs
> 
> [env:T-ETH-POE]
> board = esp32dev
> build_flags = 
> 	${env.build_flags}
> 	-DLILYGO_T_INTERNET_POE
> lib_deps = ayushsharma82/ElegantOTA@^3.1.6
> 
> [env:T-ETH-POE-PRO]
> board = esp32dev
> build_flags = 
> 	${env.build_flags}
> 	-DLILYGO_T_ETH_POE_PRO
> 	-DUSER_SETUP_LOADED
> 	-include lib/TFT_eSPI/User_Setups/Setup216_LilyGo_ETH_Pro_ESP32.h
> lib_deps = ayushsharma82/ElegantOTA@^3.1.6
> 
> [env:T-INTERNET-COM]
> board = esp32dev
> build_flags = 
> 	${env.build_flags}
> 	-DBOARD_HAS_PSRAM
> 	-DLILYGO_T_INTER_COM
> lib_deps = ayushsharma82/ElegantOTA@^3.1.6
> 
> [env:T-ETH-Lite-ESP32]
> board = esp32dev
> build_flags = 
> 	${env.build_flags}
> 	-DBOARD_HAS_PSRAM
> 	-DLILYGO_T_ETH_LITE_ESP32
> 	-DUSER_SETUP_LOADED
> 	-include lib/TFT_eSPI/User_Setups/Setup216_LilyGo_ETH_Lite_ESP32.h
> lib_deps = ayushsharma82/ElegantOTA@^3.1.6
> 
> [env:T-ETH-Lite-ESP32S3]
> lib_compat_mode = strict
> platform = espressif32@6.7.0
> framework = arduino
> 
> 
> board = esp32s3box
> build_flags = 
> 	${env.build_flags}
> 	-DBOARD_HAS_PSRAM
> 	-DLILYGO_T_ETH_LITE_ESP32S3
> 	-UARDUINO_USB_CDC_ON_BOOT
> monitor_filters = 
> 	default
> 	esp32_exception_decoder
> upload_speed = 921600
> monitor_speed = 115200
> lib_ignore = 
> 	Ethernet
> ;lib_deps = ayushsharma82/ElegantOTA@^3.1.6
> lib_ldf_mode=deep
> lib_deps =
>     me-no-dev/AsyncTCP @ 1.1.1
>     me-no-dev/ESP Async WebServer @ 1.2.3
>     arduino-libraries/Arduino_JSON @ 0.2.0
> 	ayushsharma82/ElegantOTA@^3.1.6
> board_upload.flash_size = 16MB
> board_build.partitions = partition_table.csv
> board_build.filesystem = littlefs
> [eht_elite_base]
> board = esp32s3box
> build_flags = 
> 	${env.build_flags}
> 	-DBOARD_HAS_PSRAM
> 	-DLILYGO_T_ETH_ELITE_ESP32S3
> 	-UARDUINO_USB_CDC_ON_BOOT
> monitor_filters = 
> 	default
> 	esp32_exception_decoder
> upload_speed = 921600
> monitor_speed = 115200
> lib_ignore = 
> 	Ethernet
> lib_deps = 
> 	SPI

lewisxhe, 2024-12-20:
board_build.partitions = partition_table.csv changed to board_build.partitions = default_16MB.csv

seikobu, 2024-12-20:
board_build.partitions = partition_table.csv ist not active.
I'm using huge_app.csv as per example, please see https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/platformio.ini line 92

lewisxhe, 2024-12-21:
I mean, as long as you change the partition table to board_build.partitions = default_16MB.csv, you can use OTA normally.

huge_app.csv does not have OTA partitions.

closed 2025-02-04 by github-actions[bot]