Xinyuan-LilyGO/LilyGO-T-ETH-Series #97: T-POE Pro | Ethernet Timeout - Will NOT Connect but power works.

CopperHeroin, 2025-01-07:
I think I may have a faulty board. I Have tried just about everything ChatGPT and I can think of as to why the ethernet port will not work. Wifi works but If I plug the board in via ethernet it will power on but no yaml setting combinations I can find will work and there is a surprisingly small amount of information available on this board and how to configure it. I purchased this so I could power the board with POE and use an ethernet connection so if the board is faulty I have to return it.  That being said... am I missing something here? Thanks guys!

\\YAML

esphome:
  name: esphome-web-f168ec
  friendly_name: HVAC Controller
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:

api:

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN  # Adjust as needed
  phy_addr: 0
  manual_ip:
    static_ip: 192.168.1.22
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 8.8.8.8

[image]
[image]

lewisxhe, 2025-01-08:
Test this
```
esphome:
  name: t-eth-pro
  friendly_name: t-eth-pro

esp32:
  board: esp32dev
  framework:
    type: arduino

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_OUT
  phy_addr: 0
  reset_pin: 5

  # Optional manual IP
  # manual_ip:
  #   static_ip: 192.168.36.227
  #   gateway: 192.168.36.1
  #   subnet: 255.255.255.0

# Enable logging
logger:
  level: DEBUG
  baud_rate: 115200
  logs:
    mqtt.component: DEBUG
    mqtt.client: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "dBVA+/ahpgURrxBUQ2pbJzNA9b4U/9Yk8wntuW8tJzY="

```

CopperHeroin, 2025-01-08:
It's alive!

Thank you good sir... you are a gentleman and a scholar.

The only thing it didn't like was the  "reset_pin: 5". It says it's not an ethernet attribute and to check indentation.
I moved it out of ethernet and it still didn't like it. Other than that it booted right up.

Hopefully LilyGo will work on their documentation. I am very appreciative of your help.

Thanks again!

closed 2025-02-22 by github-actions[bot]