Xinyuan-LilyGO/LilyGO-T-ETH-Series #98: IP_EVENT_STA_GOT_IP never called in eth2ap

dizcza, 2025-01-25:
I'm trying to connect an ESP32 TTGO board via T-ETH-ELITE to my router.

ESP32 TTGO  <--- softAP ---> T-ETH-ELITE <---- ethernet ----> router

For the first time, the link is successful. But once I reboot my ESP32 board, it can no longer connect to the router - no IP is assigned.

I'm using your eth2ap ESP-IDF example on T-ETH-ELITE.

ESP32 TTGO code to connect to WiFi (works fine without T-ETH-ELITE in-between):

```c

#include <stdio.h>
#include "esp_err.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_event.h"
#include "esp_log.h"
#include "netdb.h"
#include "esp_timer.h"

#include "wifi_sta.h"
#include "bsp.h"

#define WIFI_STA_STATIC_IP_ADDR        "192.168.3.2"
#define WIFI_STA_STATIC_NETMASK_ADDR   "255.255.255.0"
#define WIFI_STA_STATIC_GW_ADDR        "192.168.3.1"
#define WIFI_STA_MAIN_DNS_SERVER       WIFI_STA_STATIC_GW_ADDR
#define WIFI_STA_BACKUP_DNS_SERVER     "0.0.0.0"

#define WIFI_MAX_RETRY                 5
#define WIFI_RECONNECT_TIMEOUT_S       10

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT             BIT0
#define WIFI_FAIL_BIT                  BIT1

typedef struct {
    bool static_ip;
    esp_ip4_addr_t ipaddr;                 /* This board IP addr */
    esp_netif_t* netif;
    bool started;
    esp_timer_handle_t timer_reconnect;
} wifi_sta_t;

static wifi_sta_t m_wifi_sta = {0};

static const char *TAG = "wifi_sta";

static esp_err_t wifi_sta_set_dns_server(esp_netif_t *netif, uint32_t addr, esp_netif_dns_type_t type)
{
    if (addr && (addr != IPADDR_NONE)) {
        esp_netif_dns_info_t dns;
        dns.ip.u_addr.ip4.addr = addr;
        dns.ip.type = IPADDR_TYPE_V4;
        BSP_SOFTCHECK(esp_netif_set_dns_info(netif, type, &dns));
    }
    return ESP_OK;
}

static void wifi_sta_set_static_ip(esp_netif_t *netif)
{
    if (esp_netif_dhcpc_stop(netif) != ESP_OK) {
        ESP_LOGW(TAG, "Failed to stop dhcp client");
        return;
[... 421 lines cut ...]
dhcp_check(): set request timeout 500 msecs
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_check(netif=0x3ffcba2c) st
dhcp_check(): dhcp state is CHECKING
dhcp_check(): set request timeout 500 msecs
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): CHECKING, ARP request timed out
dhcp_bind(netif=0x3ffcba2c) st1
dhcp_bind(): t0 renewal timer 86400 secs
dhcp_bind(): set request timeout 86400000 msecs
dhcp_bind(): t1 renewal timer 43200 secs
dhcp_bind(): set request timeout 43200000 msecs
dhcp_bind(): t2 rebind timer 75600 secs
dhcp_bind(): set request timeout 75600000 msecs
dhcp_bind(): IP: 0x9403a8c0 SN: 0x00ffffff GW: 0x0103a8c0
dhcp_bind(): dhcp state is BOUND
I (20:03:17.709) esp_netif_handlers: sta ip: 192.168.3.148, mask: 255.255.255.0, gw: 192.168.3.1
I (20:03:17.717) wifi_sta: event_id 0
I (20:03:17.721) wifi_sta: IP_EVENT_STA_GOT_IP
I (20:03:17.726) wifi_sta: WiFi connection established
```

dizcza, 2025-01-25:
Rebooting T-ETH-ELITE helps.

lewisxhe, 2025-01-26:
I tested the example in the warehouse and it was normal. Now it is Chinese New Year and we are on holiday. We cannot handle your problem. We need to test it after the holiday.

The eth2ap example comes from espressif. The warehouse example only changes the chip type and pins. Please try to test it with the main line example.
https://github.com/espressif/esp-idf/tree/master/examples/network/eth2ap

dizcza, 2025-01-26:
The official example doesn't work either - if I had known it in advance, I'd have opened the issue there.

Did you test the connection after you reboot the client?

Sure, happy holiday. I may transfer the issue to the official esp-idf repo since you just borrowed their code and perhaps there's nothing wrong with the T-ETH-ELITE board.

closed 2025-03-12 by github-actions[bot]