Xinyuan-LilyGO/LilyGO-T-ETH-Series #99: T-POE-PRO, esp-idf platformio.ini settings..?

yltsa, 2025-02-03:
> This issue was closed because it has been inactive for 14 days since being marked as stale. 

 _Originally posted by @github-actions in [#94](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/issues/94#issuecomment-2617644913)_

lewisxhe, 2025-02-06:
This one ?
https://lilygo.cc/products/t-eth-elite-1?_pos=1&_sid=3a990f901&_ss=r&variant=44498204983477

lewisxhe, 2025-02-06:
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/tree/master/esp-idf-examples

yltsa, 2025-02-06:
Thanks a lot. I was trying but no success by using Visual Studio Code - Platformio - esp-idf..

lewisxhe, 2025-02-07:
This is an example of command line compilation, please see the idf quick start
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/index.html#uninstall-esp-idf

yltsa, 2025-02-07:
Hi, yes.. I have a working environment for esp-idf including Ethernet. But that is based on SPI/MOSI/MISO.. I do not know how to make PHY/Ethernet and cannot find settings for that by runing menuconfig. The SPI/Ethernet pins are supported.

lewisxhe, 2025-02-07:
https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/18e1e552e7de71d489640faa6a7109f0bd63bcb2/esp-idf-examples/basic/sdkconfig.defaults#L539

yltsa, 2025-02-07:
Hi, I do not understand that. T-POE-PRO does not use SPI Ethernet.. or? There are three pins: RESET, MDC, MDIO. Need configuration for those..?

[image]

lewisxhe, 2025-02-08:
Are you talking about this board? There is no idf example for this board. This esp official example can be used after changing the GPIO
https://github.com/espressif/esp-idf/tree/master/examples/ethernet

closed 2025-03-24 by github-actions[bot]