Xinyuan-LilyGO/LilyGO-T-ETH-Series #100: T-ETH ELite -> PIN layout wrong (IO9 and IO11)

pixelwave, 2025-02-06:
I think I found an issue with the shown pin layout of the T-ETH ELite board. On this image IO9 and IO11 are actually the other way around:

[image]

In ESPHome I tried to configure it using this pin assignment:

```
# DISPLAY 
LCD_DIN: GPIO9 # MOSI
LCD_CLK: GPIO10
LCD_CS: GPIO40
LCD_DC: GPIO41
LCD_RST: GPIO13
LCD_BL: GPIO2
```

But I got no image. Only when I changed (physically!) and by accident PIN IO9 and IO11.

lewisxhe, 2025-02-07:
[image]
You're right, I'll report this to the guy who made this image.

pixelwave, 2025-02-07:
Awesome! Thanks a lot hopefully it will safe other people a ton of troubleshooting time.

closed 2025-03-24 by github-actions[bot]