Xinyuan-LilyGO/LilyGO-T-ETH-Series #104: T-ETH-Lite ESP32 (Not the S3) SPI/I2C question

BliXem1, 2025-06-11:
Hi,

I want to buy this one with the POE addon. But I want to run some I2c devices + one AS3935 SPI device on this ESP. Is that possible to run simultaneously?

I tried another company with W5500 chip on S3, but that won't work, it won't see the 'lightnings'.

I have some I2c extension board so I can plug them on that one for more then one device.

Let me know if possible.

lewisxhe, 2025-06-11:
Of course you can.
ESP has 3 available SPIs, one of which is allocated to FLASH
One is allocated to W5500, and the other is free for you to use
I2C can also use two, there is no fixed GPIO, you can specify the GPIO during initialization, please see the examples provided in the examples

BliXem1, 2025-06-11:
Hi @lewisxhe thanks for the quick reply.

Weird that it didn't work with my other board. Anyways, I'm not gonna use the S3 version but only the ESP32-WOVER-E with POE. But I see you are talking about the S3 version, it's still possible with ESP32-WOVER-E? So the H673 version: https://lilygo.cc/products/t-eth-lite?variant=43120880746677.

Please confirm, thanks :)

lewisxhe, 2025-06-11:
Yes. They are basically the same. But the ESP32 version has a lot less IO.

lewisxhe, 2025-06-11:
[image]

BliXem1, 2025-06-11:
Yeah i did see that, but it did confuse me with I2c because of this:

`#elif defined(LILYGO_T_ETH_LITE_ESP32)
// IO35,36,37,38,39 can only be used for input and cannot be set as output
#define I2C_SDA     13
#define I2C_SCL     14`

Which is on the same GPIO's. But you did say that I can choose any GPIO which are not in conflict ofc. So I will try that too.

lewisxhe, 2025-06-11:
SDA:21 SCL:22

BliXem1, 2025-06-12:
Alright! Thanks for the replies.

closed 2025-06-12 by BliXem1

BliXem1, 2025-06-13:
Alright, got a S3 version and tested it out with my AS3935 and via POE. Problem is that I don't get any lightning detection when I test it out. Without POE on the same GPIO it works.

I know that it's not your problem but what do you think what I can test to make it work via POE? I'm running on ESPhome btw.

BliXem1, 2025-06-16:
SPI, and the GPIO's of your screenshot. Via POE not working, no lightning detection, and via USB (TTL) it's working. Weird heh? Also, the shield POE does the same, not working. So I think there is to much noice.

BliXem1, 2025-06-16:
You mean the AS3935 module, then no, i didn't check that. But I tested it with something else, and thats a temperature module, that did work via i2c.

lewisxhe, 2025-06-16:
Then other modules should work properly.
How did you connect them? Can you describe it?

lewisxhe, 2025-06-16:
Or measure whether your external module has voltage

BliXem1, 2025-06-16:
Via 3v3, gnd, and the SPI ports, IRQ on GPIO04. Ofc the as3935 module. I will test it later. For now I use a ESP32-S3 with USB-C. I think it's the noice on the line, because via USB TTL it's working without POE ofc.